MyRobotLab/myrobotlab #140: ProgramAbGui - Reload Session button

supertick, 2017-11-04:
Ray says,

> Small bug with the ProgramAB service GUI. I'm launching and starting the service using a python script, however the GUI still has the Start Session button instead of the Reload Session button

moz4r, 2017-12-13:
Fixed

Sebchko, 2017-12-17:
Hello! Just to say that the reload button don't work on my config.
(before the update, this button was labeled "start session" at first but the text changed to "reload" after the first use. )
@+

moz4r, 2017-12-17:
Hi if you change the username this will reload the session with the new username. But maybe we need a restart buton

Sebchko, 2017-12-17:
Don't know if we need it, but I was using it :)
, I make a vocal comment to reload the session so ... no worries
(restarting MRL even in "light mode" to test any small change in the aiml is really a pain in the ass)

moz4r, 2017-12-17:
you right , a raw restart will be useful, few mods pushed : 

restart chatbot : just restart engine
restart w/o aimlif : restart by using aiml at startup ( useful for real time debug )
reload session : change username ( for predicates )

no more troubles in the ass ahahaha

closed 2017-12-21 by moz4r