MyRobotLab/myrobotlab #141: Augment ImageDisplay with more functionality

supertick, 2017-11-06:
- overlay
- transparencies
- merging video streams
- animated gif + videostream
- addText
- updateText

Initially I started an OpenCV filter called Overlay.  But OpenCV is lacking in a lot of aspects that Java works well in.  Specifically, merging transparencies cannot be done without a bug work-around in opencv, but it is straight forward in "pure" Java without the JNI

I was thinking of adding overlays to our ImageDisplay service, so that it would work with a video stream, full screen.

The interface perhaps for blending images together could possibly mimic Processing's [blend](https://processing.org/reference/blend_.html) interface.

Later we might consider using a pure Java image processing library like [ImageJ](https://github.com/imagej/tutorials) or [ImageJ2](https://imagej.net/ImageJ2)

References 
-  [How to blend images in Java](https://stackoverflow.com/questions/13425797/overlay-a-transparent-color-on-top-of-a-png-with-alpha-channel)
- [ImageJ Jython examples](https://nbviewer.jupyter.org/github/imagej/tutorials/blob/master/notebooks/ImageJ_Tutorials_and_Demo.ipynb)
- [Programming Guide for ImageJ](http://albert.rierol.net/imagej_programming_tutorials.html#ImageJ)
_ [ImageJ examples in Jython](http://imagej.net/Jython_Scripting_Examples)

kwatters, 2018-09-25:
Move to Ogre ?

supertick, 2018-09-25:
yup