MyRobotLab/myrobotlab #144: Virtual InMoov Not Working

Ashish-05, 2017-12-09:
I am currently running MRL 1.0.2652 on Windows 7 64-bit OS. I am facing issues with Virtual InMoov. When MRL starts, I hear the message,"Alert ! Virtual inmoov cannot start, missing j. m. 3. folder or , you do not have a java 64 bit version." I have Java 8 64-bit installed and the jm3 folder is present in C:\MRL\InMoov. 

On asking around a bit, I came to know that there are some recent versions that have an issue with Virtual InMoov, because they are beta versions. I could try myrobotlab.1.0.2340.zip, but apparently this versions has issues with Face Tracking which is an integral part of my use case.

I have Arduino, Chrome Java 8 installed as suggested in the MRL install instructions. Are there any other dependencies I should install?

I would like to get started with Virtual InMoov, but since I want something simple which is suitable for layman users, I want to avoid going through the Eclipse, Blender route. 

This link https://github.com/MyRobotLab/myrobotlab/releases only lists versions upto 1.0.1758. Where can I see the rest of the versions? I want to compare and decide which version would be suitable.

I have already sent a No-worky. Kindly help me out, How should I proceed?

supertick, 2017-12-09:
I suspect you installed 64 bit Java 8 in the browser and from a command shell still have 32 bit Java installed. 
To verify this you start a command shell and type 
```
C:\>java -version
java version "1.8.0_101"
Java(TM) SE Runtime Environment (build 1.8.0_101-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.101-b13, mixed mode)
```
if 64-Bit does not display for the result, you have 32 bit and need to install Java 8 64 bit on the operating system (not the browser)

For the latest go to http://myrobotlab.org ... top left there is a link which says "latest build"

juergmaier, 2017-12-11:
Hi Grog
Thanks, I get this output from java -version, so it looks to be 64 bit?

C:\Users\marvin>java -version
java version "1.8.0_151"
Java(TM) SE Runtime Environment (build 1.8.0_151-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.151-b12, mixed mode)

Juerg

Ashish-05, 2017-12-13:
Thanks for your reply GroG. You were correct.
At first, I downloaded the 64-bit Java 4 Update 151 from this [link](http://www.oracle.com/technetwork/java/javase/downloads/jre8-downloads-2133155.html) and installed it following the instructions provided [here](https://java.com/en/download/faq/java_win64bit.xml). But, it did not  work.

I then first uninstalled the Java Runtime Enviornment and JDK from my PC completely, and then installed the [update](https://java.com/en/download/manual.jsp) again. This worked.

I then installed MRL 1.0.2661 and after downloading all the components, shutdown MRL and then started the START_INMOOV.bat file. This time, Virtual InMoov started! Progress! But, I am not able to control it with the Start Left Hand, Start Right Hand, Start Head etc buttons in the "i01" tab or the servo sliders.Also, it does not move its jaw when I talk to it, though I am able to . So, the Speech Recognition, ProgramAB and MarySpeech services are working. 

I am now trying to get Virtual InMoov to function and move its various joints like it would in the physical world(that is Virtual InMoov's purpose after all).

Thanks for your help GroG!

hairygael, 2017-12-13:
Hello,
It's great that progress is done!
If you have used START_INMOOV.bat (without modifications to the config
files) to launch Virtual InMoov, you can use the sliders in each skeleton
tabs to move parts of the robot. You do not need to use Start Left Hand,
Start Right Hand, Start Head..
Note: the Virtual fingers aren't implemented.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Ashish-05, 2017-12-13:
I have been trying to use the sliders but Virtual InMoov does not move. When I click in the Virtual InMoov window, the 3D model rotates and the Frames per Seconds shows values and keeps fluctuating between 2-10 fps. 
Other than that, nothing.
Thanks for your help, Sir!

hairygael, 2017-12-14:
Very good progress!
But now you say the sliders aren't actuating the Virtual InMoov?
Did you by any attempt modify the config files?
Make sure your InMoov/config/*service_F_VirtualInmoov.config.default* is
set like this:

virtualInmoovAlwaysActivated=False
;use real arduino + virtual inmoov
VinmoovMonitorActivated=False
;some kind of control center, unfinished
ForceVinmoovDisable=False
;never launch JME3/Virtual Inmoov

Also make sure, your InMoov/config/*_InMoov.config.default* is set like this:

ScriptType=Virtual
; RightSide: Also called FINGERSTARTER : connect one arduino ( called
right ) to use FingerStarter + inmoov right side
; LeftSide: connect one arduino ( called left) to use head / inmoov left side
; setup your com ports inside service_6_Arduino.config
; NoArduino: vocal Only
; Full: Both side arduinos connected
; Virtual: virtual arduino and inmoov !
debug=False

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Ashish-05, 2017-12-14:
Yes Gael, the files are as you say they should be. I haven't changed any of the config or any other files.
Thank You for your help and guidance!

hairygael, 2017-12-14:
Mmmh, I must confess that you are the first one to report this pb, and I
can't see what would be wrong.
I would guess it is related to your previous wrong installation, something
must remain.

I suggest, you come over on MyRobotLab and create a post reporting your
issue by copy/paste the conversation we had here.
And send a No Worky via the help button (top right of MRL SwingGui window)
Our fantastic developpers will certainly sort it out!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Ashish-05, 2017-12-14:
I tried installing MRL on another laptop.Virtual InMoov is working fine! I guess my system's wasn't upto the task in terms of its graphics capabilities. My system was only giving me 2-10 fps whereas the other system reaches 60 fps.
Thanks to everyone for your patience and support.

closed 2017-12-14 by Ashish-05