MyRobotLab/myrobotlab #145: Issue with NaturalReaderSpeech mixed up voices

CheekyMonkey, 2017-12-13:
Issue with NaturalReaderSpeech mixed up voices

Commentary here for reference : https://github.com/MyRobotLab/myrobotlab/commit/d7ae81c867c14465752cd36b0d856d365e91dcad

Python test script [here](https://github.com/MyRobotLab/pyrobotlab/blob/develop/home/CheekyMonkey/naturalreaderspeech-v2-test.py)

Reference video [here](https://www.youtube.com/watch?v=JpMQ6OwDyTw&feature=youtu.be)

[image]

hairygael, 2017-12-13:
> Commentary here for reference : d7ae81c

As you can see I tried to reformate the correct name list with numbers in a
logic order in a attempt to fix the mixed up voices, but that didn't do the
trick.
I have also tried to find on NaturalReaderSpeech site the liste with
numbers which would indicate the correct numbers, but my knowledge in that
field has limits.
Hopefully a charitable soul will come to rescue.
:)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

CheekyMonkey, 2017-12-13:
Thanks Gael, let's wait and see :)

moz4r, 2017-12-16:
charitable soul in progress

hairygael, 2017-12-16:
:)
Keep me informed!
Because for now when I want the Ryan voice, I get Tim's voice instead (I'm
not even sure it's really Tim, though)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-12-16:
dig into code, it if funny natural use 2 external providers. Amazon Polly and .. Ibm Watson . I let you know

moz4r, 2017-12-16:
Hacked, this message will autodestruct .
Worky ( still no guaranties about longevity, but no magic here, external providers used by natural, we can borg them later if it not done already ) , 

http://myrobotlab.org/service/NaturalReaderSpeech?branch=develop

CheekyMonkey, 2017-12-16:
Hi moz4r

Seems still noworky, for me at least (build 2673 )
All voices appear to return the same male voice

test script [here](https://github.com/MyRobotLab/pyrobotlab/blob/develop/home/CheekyMonkey/naturalreaderspeech-v2-test.py)

closed 2017-12-21 by moz4r