MyRobotLab/myrobotlab #146: connection problems between google talk and myrobotlab chat

odinroy666, 2017-12-16:
hello, I need help with the immov's speech, when I start the system everything seems to be going well, I have configured Spanish language, and speaks perfectly, opens the page in goggle without oakwoods and recognizes the words well, on the other hand, myrobotlab runs also, but although the chatbot works, it does not recognize what I say, it only speaks when I press chat button, and it does not execute orders and it does not even answer that it does not understand them. besides that the mouth moves too slowly. Thank you very much in advance, and I hope you can help me out.

hairygael, 2017-12-16:
Hello,
Which version of MyRobotLab are you using?
It seems you maybe have the version 2340 or prior, if yes, you need to
press the buttons in WebKitSpeech recognition.
Auto Listen On and Speedup recognition.

More important, if you have the version 2340, the Spanish brain wasn't able
to do much, and if I recall the gestures were not implemented.
The brain in Spanish isn't very developped because the Spanish community
hasn't really improved it yet.
We need Spanish help!!

To get the latest MyRobotLab and InMoov version, which should work with the
gestures, here is the link:
http://34.201.4.170/deploy/develop/myrobotlab.jar

<http://34.201.4.170/deploy/develop/myrobotlab.jar>

<http://34.201.4.170/deploy/develop/myrobotlab.jar>
To install wiki:
https://github.com/MyRobotLab/inmoov/wiki/HOWTO---SETUP-&-PREREQUISITES

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

odinroy666, 2017-12-16:
Ok, i try tomorrow, and when i learn some more to use myrobotlab and phyton
i try to do something for spanish comands, translate or another thing, but
today i dont know how help. 😬. Thanks for the help. When i can, i
translate the orders to spanish and send for the comunity for correct the
scrips.

odinroy666, 2017-12-18:
Effectively, it works, update myrobotlab and start, now we need to sharpen the servos a little, get the eye camera working, and finish the arms, thank you very much Gael. If you want and although I'm not a great expert, if you tell me what scripts are what I need to change to be able to use Spanish I will get to work so that they are finally in Spanish for future builders, and so be able to help. I'll be really happy. Thanks and regards

De: Langevin Gael
Enviado: domingo, 17 de diciembre de 2017 0:51
Para: MyRobotLab/myrobotlab
CC: odinroy666; Author
Asunto: Re: [MyRobotLab/myrobotlab] connection problems between google talkand myrobotlab chat (#146)

Hello,
Which version of MyRobotLab are you using?
It seems you maybe have the version 2340 or prior, if yes, you need to
press the buttons in WebKitSpeech recognition.
Auto Listen On and Speedup recognition.

More important, if you have the version 2340, the Spanish brain wasn't able
to do much, and if I recall the gestures were not implemented.
The brain in Spanish isn't very developped because the Spanish community
hasn't really improved it yet.
We need Spanish help!!

To get the latest MyRobotLab and InMoov version, which should work with the
gestures, here is the link:
http://34.201.4.170/deploy/develop/myrobotlab.jar

<http://34.201.4.170/deploy/develop/myrobotlab.jar>

<http://34.201.4.170/deploy/develop/myrobotlab.jar>
To install wiki:
https://github.com/MyRobotLab/inmoov/wiki/HOWTO---SETUP-&-PREREQUISITES

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-12-19:
Great to hear it's working!
You can find the Spanish brain of InMoov in your directory:
mrl/InMoov/chatbot/bots/es/*aiml*/

On Github in the develop branch (beta):
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/chatbot/bots/es/aiml

This is the English version which contains the current best brain:
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/chatbot/bots/en

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

odinroy666, 2017-12-19:
OK, well, I'll look at the brain in English that does not have the brain in
Spanish and translate the missing files, then I'll try them and I'll tell
you. Thanks again

odinroy666, 2017-12-19:
hello again, I have started to translate into Spanish, the files contained in the bots folder, there are many missing files that do exist in folders like "fr" or "en", if I want to add some, could I copy it directly from one of those folders to the folder "is" or I need to change some more file. Greetings and thank you
P.S. When I have them, how can I send them to be used by whoever they want?

De: Langevin Gael
Enviado: martes, 19 de diciembre de 2017 10:00
Para: MyRobotLab/myrobotlab
CC: odinroy666; Author
Asunto: Re: [MyRobotLab/myrobotlab] connection problems between google talkand myrobotlab chat (#146)

Great to hear it's working!
You can find the Spanish brain of InMoov in your directory:
mrl/InMoov/chatbot/bots/es/*aiml*/

On Github in the develop branch (beta):
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/chatbot/bots/es/aiml

This is the English version which contains the current best brain:
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/chatbot/bots/en

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

kwatters, 2018-09-25:
can we close this issue? or move it to the ogre release?

wilcovantoorn, 2018-09-25:
In my opinion this is a translation issue, and i dont know how far the
starter got. We could move it to ogre to see if it is updated enough that
people can use it. If not we can ask a spanish builder to help with
translation. I know dutch language is done by gael in the most basic way,
so my take on it is we move it to ogre and in a translation to-do. It wasnt
a bug as such, more an enhancement.

odinroy666, 2018-09-25:
Hi, i dont understand the last post, actually works with two people to translate the inmoov script to spanish, i send mail to gael but i think he dont read, i start only and is hard work for only one person, now i find another spanish maker and he wants to help me to translate, but we go slowly because we have another works. When we have more things i tell in github. Thanks

wilcovantoorn, 2018-09-25:
Because nixie is in final stage before deploy, we need to look at what will
be pushed to ogre, the next version and what is ready now.
We are very glad with your help translating to Spanish. Because it is so
much work it would be better to move your ticket (translation) to ogre, so
we can focus on bug hunting in nixie.
It is not that we are not stopping and closing your github ticket but only
moving it to the next version.

odinroy666, 2018-09-25:
ok great, it seems perfect, I already have some translation but I was very stopped by work issues, and soon I'll get back to it, and also with help, so I go faster, since I have the immov in half, and I also need to continue with it, as soon as I have more advanced I will raise it to encourage more Spanish developers, that with inmoov I do not find many, it took me almost a year to find two makers and from provinces different from mine, if you move the thread I will Attentive not to lose it, and be able to continue it later because I do not want to lose contacts. Thank you very much and luck.

supertick, 2022-07-09:
There is a chatbot Spanish brain
Additionally there is a Azure Translator service

closed 2022-07-09 by supertick