MyRobotLab/myrobotlab #148: Neopixel service ERROR c.myrobotlab.framework.Service - InvocationTargetException

CheekyMonkey, 2017-12-20:
When running the neopixel ring [test script](https://github.com/MyRobotLab/pyrobotlab/blob/master/service/NeoPixel.py), I observe an exception being thrown in the mrl logs

Noworky log sent  (20/122017 22:00 GMT)
Exception attached for reference

It's worth pointing out that the neopixel ring runs through it's animations without any issue despite the exception being thrown

Setup:
Raspberry Pi 3
Raspbian Stretch
Arduino Nano V3
myrobotlab build 2679
http://myrobotlab.org/service/NeoPixel

[mrl 2679 neopixel exception.txt](https://github.com/MyRobotLab/myrobotlab/files/1577160/mrl.2679.neopixel.exception.txt)

moz4r, 2017-12-21:
Hi ! can you push the complete script u use to test ? seem SwinGui neopixel need a little fix

CheekyMonkey, 2017-12-22:
Hi moz4r

The python test script used is [here](https://github.com/MyRobotLab/pyrobotlab/blob/develop/home/CheekyMonkey/neopixel%20script%20usb%20connection.py)

moz4r, 2017-12-23:
little fix pushed , I hope it is better. it is a strange issue related to swing update...

CheekyMonkey, 2017-12-26:
Thanks moz4r, nice work.

Observations so far as at build 2690:

- The exception is no longer thrown as an "ERROR"
- The exception now appears as a repeating "WARN" entry in the mrl application log 

Is the "WARN" log entry to be expected? 

[image]