MyRobotLab/myrobotlab #153: MyRobotLab moves to Maven

supertick, 2017-12-31:
Make necessary changes to enable Maven builds.  

Manage dependencies via Maven & new MyRobotLab Artifactory service.

Change directory structure to align with Maven conventions.

Make Jenkins Pipeline build & deploy new snapshot build from develop branch

Change runtime dependency embedded Ivy to handle new dependencies

supertick, 2018-01-17:
Kwatter and I tag-teamed this !

Make necessary changes to enable Maven builds.

~~Manage dependencies via Maven & new MyRobotLab Artifactory service.~~

~~Change directory structure to align with Maven conventions.~~

Make Jenkins Pipeline build & deploy new snapshot build from develop branch

~~Change runtime dependency embedded Ivy to handle new dependencies~~

kwatters, 2018-03-06:
The develop branch now uses Maven to run in eclipse, and maven can also produce a build, however, runtime installation of services and their dependencies is still no worky.

supertick, 2018-04-08:
Works ! we also have the "latest" link hooked up and a version number from Jenkins !
Starting with 1.1.37 -  Rockn !

closed 2018-04-08 by supertick

hairygael, 2018-04-10:
> runtime of services worky & pipeline deploy new snapshot to "latest" link
> worky !

Great work!!
I have just downloaded the latest MRL and will test.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>