MyRobotLab/myrobotlab #154: 1 Speech service to rule them all

moz4r, 2018-01-03:
Idea : 
- Merge all speech services code as possible ( I understood we have interface & abstract )
- New UI screen with cool combo filters ,test voice button, etc ... 
- Shared audioFiles voices effect for robot expressions

Question:
For some engines we can store API keys Inside the aes safe service. I also heard a mrl cloud server++ can be implemented. What is the direction for this ? 

[image]

Todo: 
Adapt marySpeech to use audioFile if possible.
...

supertick, 2018-01-04:
- Different cloud speech synthesizers have different apis, the interfaces create a common interface for all "consumers" of the speech synthesizers, and I created the AbstractSpeechSynthesis as a class which implement common & shared methods for "ALL" speech synthesis services.
AcapelaSpeech, IndianTts, LocalSpeech, MarySpeech, MicrosoftSpeech, MimicSpeech, NaturalReaderSpeech, Polly, & VoiceRss now all derive from AbstractSpeechSynthesis (goodtimes) .. so what merging are you interested in ?  (this is goodtimes :)
- The UI should follow the same pattern as the Abstract, where there is one common shared cool UI with buttons and such, and they all inherit or derive from it.
- "Shared audioFiles voices effect for robot expressions" sounds good, not sure what exactly you have in mind though .. need more details !

MyRobotLab Cloud Service - Yes, very interested in it - I think it would be a benefit to many, if we pooled resources at a cloud level, and was able to simplify keys and reduce costs through potentially securely sharing  many keys being being utilized by a single mrl key.  Additionally, caching @ mrl & locally would have advantages.  Regardless of those details, we all want our robots to be able to talk to one another .. so ya mrl cloud should blossom this release ...

We currently have a new "medium" server serving the "repo" for all the dependencies not in maven central or other repos.  This is the biggest resource we have had and I think it would be a good starting place for mrl cloud services

moz4r, 2018-01-04:
Idea is to provide a unique abstract speech synthesis service ( with backward compatibility ) . I think we can put more shared code Inside AbstractSpeechSynthesis.

```
speech=runtime.start("speech","Speech")
speech.setLanguage("EN") # filter voices & engines by language & OS detection
# By default only offline engines will be available.
# speech.setMrlKey("xxx") ->  online & nofree voices ( cached on server+client )
print speech.getVoices() # return available voices ( engines merged )
```

About "Shared audioFiles voices effect" it will be great if bot can laugh or cry. And I remember some tags Inside Acapela speech. Goal is speech service intercept the standardized tag and call the corresponding audioFile effect. 

`speak("Oh ! that is so funny #LAUGH01# ") #LAUGH01# tag will call LAUGH01.mp3 binarie`

Maybe those voice effects can be pushed to cloud service instead of locally. So If mrlKey is not set, #LAUGH01# return empty.

Yeh this new tool will offer some exciting functionalities!

supertick, 2018-01-05:
Abstracts cannot be actual services ...  so I think your saying a single Speech service which handles a composite of all others ?  Is that correct ?

So new Speech service would just call other real speech services like MarySpeech, Acapela etc.  
It would have configurable ordering ?  So that people could change the order of which "real" speech service would be tried if the preceding one failed ?

Pulling more code out of the services and putting it in AbstractSpeechService is good for everyone.. even if I did misunderstand you ...  although you don't want to "bloat" it as a composite ..  a good rule of thumb is if you see the same code in 3 services it definitely should be "pushed up" into  AbstractSpeechService 

Ya I remember the Acapella emotions .. I'm all for it .. the best place for it would probably be in the Artifactory server .. (same place where all the "custom" jars like ProgramAB come from now)

moz4r, 2018-05-14:
Seem we are close to close this ticket.. 
Also voices emotions pushed but we need to find a final place for them. I'm using my remote server to store them for now. Just pushed few . 

https://www.youtube.com/watch?v=drS550g2nfQ

http://www.myai.cloud/mrl/audio/

hairygael, 2018-05-16:
:)
Nice

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2018-05-27 by moz4r