MyRobotLab/myrobotlab #155: HttpClient refactoring - use Mrl Peer Service

supertick, 2018-01-07:
there is a lot of these dependency definitions in a variety of services :

meta.addDependency("org.apache.httpcomponents", "httpclient", "4.5.2");
meta.addDependency("org.apache.httpcomponents", "httpcore", "4.4.6"); 

To normalize all the dependencies we should be adding Mrl's HttpClient as a "Peer" dependency ...  I know Peers are to be depreciated in a lot of places, but in this area (describing dependencies) I think they are useful.

If all Services used Mrl's HttpClient Service, the service would evolve more quickly, and be more useful, additionally there would be only 1 place to manage versions of httpcore !

kwatters, 2018-09-05:
I think apache http client should just be part of the shaded dependencies  (in the myrobotlab.jar) and it should be excluded from everywhere else..  

this is probably still a good candidate for nixie as it's really related to the cut over to maven.

supertick, 2018-10-07:
Fixed - added dependencies to Runtime

closed 2018-10-07 by supertick