MyRobotLab/myrobotlab #157: IntegratedMovement - dependencies or service name is incorrect

supertick, 2018-01-09:
The IntegratedMovement service has dependencies of InMoov - if its a service only for InMoov it should have InMoov dependencies and be renamed InMoovIntegratedMovement.  If its to be a more general service, then the code needs to be refactored and the dependencies dropped.

hairygael, 2018-01-09:
Mmmh,
Hopefully a nice Elve is going to help on that because I wouldn't know
where to start!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2018-01-10:
Calamity was the awesome wizard who created it ...
So, this was to give him a chance to decide which path to take ..
If he's too busy, then I'll probably just rename it to InMoovIntegratedMovement

Calamity01, 2019-07-21:
dependency for inMoov should be removed as integratedMovement is not just for inMoov and don't use inMoov in his code. I don't remember why it was added,

closed 2019-07-21 by Calamity01