MyRobotLab/myrobotlab #162: Tag Word added to Abstract SpeechRecognizer

supertick, 2018-01-22:
A "Tag" word which changes the context the robot so that it knows its in an "attentive" state.   The duration of the state could "timeout" if no recognized command or activity is found.  

It's certainly more inline with the popular "Personal Assistant" applications such as Alexa & Google Home.

See more details here :
http://myrobotlab.org/content/voice-command-trigger-robot-name-only

kwatters, 2018-12-16:
I think @moz4r   added some stuff for this with the lock out grammar commit...  So this might be done?  I'm not sure.

supertick, 2019-01-03:
No its not done.  There is a time involved and it needs to be done consistently the same in a speech recognizers (ie it needs to be done in AbstractSpeechRecognizer)

Moving to Ogre..

supertick, 2022-07-09:
This was implemented in AbstractSpeechRecognizer
Its called WakeWord

closed 2022-07-09 by supertick