MyRobotLab/myrobotlab #163: Mouthcontrol Sync issue

DSBRobotics, 2018-02-09:
MOUTHCONTROLJAWTWEAK=True
Will not continue loading MRL.

supertick, 2018-02-09:
@DSBRobotics  - do you send us a noWorky ?

DSBRobotics, 2018-02-09:
I have, twice. One my tablet and one for my laptop.

DSBRobotics, 2018-02-16:
Any ideas or do you need more info?

moz4r, 2018-02-16:
Hi DSB ! There is a little bug, please look at this line Inside head.py file. 

https://github.com/MyRobotLab/inmoov/blob/master/InMoov/skeleton/head.py#L127

Just change set**D**elays by setdelays

DSBRobotics, 2018-02-16:
Need a little more explanation, I’m not a programmer, working on it. So do I change “.setDelays” to “setdelayes “ or add??
Thanks for your patience

hairygael, 2018-02-17:
> skeleton/head.py#L127

Thanks Anthony for spoting this bug!!
@DSRobotics:
Replace "setDelays" by "setdelays".

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

DSBRobotics, 2018-02-19:
Thanks for helping. Correction did allow program to continue and load but did not fix delay in mouth movements. Now as talking just holds mouth open, no movement. Tired changing sets but nothing yet has helped.

Scott

moz4r, 2018-02-20:
Hi scott, maybe your jaw servo is not well calibrated : 
Can you check this : 
Move Jaw servo to 0 -> Jaw must close
Move jaw sevro to 180 -> jaw must open 
Is it inverted ?

DSBRobotics, 2018-02-20:
It doesn’t move from 0-180 but it works well in older program. Same settings used. My issue is trying to delay movement to sync with audio. Using delay now only makes full movement, so he just opens mouth, like saying “duh”

DSBRobotics, 2018-02-20:
To correct Jaw “input MAP” is 0-180... output MAP 37-85. Fully closed at 0, fully open at 180.

Scott

moz4r, 2018-02-20:
Ok your calibration is good ! so I understood you want to sync jaw with audio, instead of software mouth control ?

DSBRobotics, 2018-02-20:
There is a jaw tweak command that I was trying to use. I can go the audio route but was trying to keep it simple but maybe that’s really the simple way at this point

DSBRobotics, 2018-02-27:
Still have issue syncing mouth movement with speech. Maybe it is how I’m trying to use it. Can someone please explain how “MouthControlJawTweak” works!! 
State to false , jaw moves well before speech. I make statement “true” and jaw just opens wide with speech but still not sync.

supertick, 2022-02-06:
UI shows configurable values - user can adjust to his/her preference 
[image]

closed 2022-02-06 by supertick