MyRobotLab/myrobotlab #164: Servo calibration

moz4r, 2018-02-27:
Today if we want to calibrate a servo ( ex. for inmoov ), it's not clear enough for everyone.
- Launch servo service
- Connect to controller
- Play with cursor
- Store limits inside config file / diy script

I think we can use existing servo UI and use existing json related to store values. ( and it's almost worky ) . Better for real-time adjustments.

hairygael, 2018-03-01:
Yes this needs to be more clear.
Too many users fail or do not understand how to calibrate a servo to report
the min and max values in their config files.
In the end they keep the default values that are preset in the config files
and might burn their servo.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

kwatters, 2018-03-06:
I suggest moving all of this calibration into java land.  We should add a button to the InMoov for "load" and "save" calibration.  This way users can use the GUI , set their min/max , com ports, velocity.. etc.. and then click save .  No more human editing of files.

moz4r, 2018-03-06:
what ! No more human editing ? I like notepad++ :)
yep agree... I think of 2 buttons
save ( not really a button this is/will be automatic ) -> store stuff inside service related json at shutdown based on py commands or gui changes. But maybe it will be safer to add a save button.
export -> this will create a thing to implement inside a diy script ( this is not mandatory because diy script will automatic load previous settings stored inside servoX.json ( if there is no override commands ), so useful to override every time )

hairygael, 2018-03-06:
The issue for beginners is not the "saving" but to understand what is the
goal of calibration and how to proceed.
Sebastien had done a very nice graphical proposition to add to the InMoov
site for beginners, but we kind of stopped because it would be better to
implement such graphic calibrator  directly on MyRobotLab.

Have a look at what EZ has done ( using my picture without permission, huh!)
https://www.youtube.com/watch?v=ECWUPJ0oDaQ&t=26s

Could we do something similar since i have the permissions to use my
pictures, ehehe

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-03-06:
The save/export calibrate will remove the actual text edit step. Just 1 thing less to do.
I think we have almost all pieces for fireworks, the core of servo calibration process will be inside servo gui screen. Maybe we can polish more servo Gui . 
Ideas welcome like 2 tabs inside servo GUI ( basics / advanced )...
WIP to do similar thing inside i02 ( swing & layout things can cause headaches, example fit all screen size with a picture like this ) . We will implement something closer, will post soon a proposition to talk about. not yet arrived at the skeleton part :)
[image]

hairygael, 2018-03-07:
Woo, that seems to be nice and promising!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-05-21:
cooked ! http://myrobotlab.org/content/servo-calibration
This is generic calibration process inside mrl , a sub part for next works for inmoov

closed 2018-06-07 by moz4r