MyRobotLab/myrobotlab #166: Python tab freezes

scognito, 2018-03-02:
Hi,
thanks for the awesome work with MRL.
I noticed that that the python tab freezes often, so my development process is very slow and stressful.
Also the lack of some basic functions like find/replace, and the prompt for overwriting exisiting file on save make it even more difficult.

Is there a chance to run python scripts without using myrobotlab? Maybe importing libraries and such.
Would be awesome, then I can use myrobotlab when my code is stable enough.

Thanks

juergmaier, 2018-03-03:
Hi Scognito

The way I work with python and mrl is using rpyc to send messages from a
python "def" section in MRL and control the robot through it's REST api.
This allows also to run MRL and the python task on different machines. Take
care to use the same python version (2.x, 3.x) and architecture (32/64
bits) on both machines.

I use the free Visual Studio 2017 on my desktop windows machine and run MRL
on a laptop mounted on my InMoov cart. Of course that python task can also
run on the laptop once you are happy with it.

If you are interested in more details of the setup drop me a note.

Juerg

juergmaier, 2018-03-03:
> Yes I'm interested, even if my tablet is 32 bit and my pc is 64. I'll find
> a solution (VM?). Can you share some information?

Give this a try:

on your pc (64 bit) you can run vs 2017 and select in the project plane the
python environment to use. Microsoft has a nice tutorial how to change the
python version in the project. I use Anaconda as it comes with lots of libs
preinstalled.

for making rpyc available to MRL download the zip from
https://pypi.python.org/pypi/rpyc/3.2.3, unzip it and copy the whole rpyc
folder into my MRL "pythonModules" folder. RESTART MRL to make it see it!

In MRL:

def callRemotePythonMethod():

  # needs unzipped rpyc folder in mrl/pythonModules

  # and a restart of mrl

  import rpyc

  # use of localhost or IP address of machine running the

  # receiving python task. This task needs to run the rpyc listener

  # on the referenced port (20001) and it must be running

  connRemote = rpyc.connect("localhost", 20001)

  # call the exposed function of the python task e.g.

  orientation = connRemote.root.getOrientation()

In Python task:

import rpyc

CART_PORT = 20001

class cartCommands(rpyc.Service):

    def exposed_getOrientation(self):

        return 123

if __name__ == '__main__':

    from rpyc.utils.server import ThreadedServer

    t = ThreadedServer(cartCommands, port=CART_PORT)

    t.start()

supertick, 2022-02-06:
swing ui no longer supported - webgui has asynchronous updates, so long scripts should not freeze ui

closed 2022-02-06 by supertick