MyRobotLab/myrobotlab #167: movement of non-linear motor servos

regirage, 2018-03-07:
Is it possible to have servo rotation with different speeds when it is running? how for example a reduced speed at the end and at the beginning

moz4r, 2018-03-07:
ohh dude you want setAcceleration and setDeceleration that will be awesome. @Calamity01 started to work on it few time ago, will keep an eye on this ticket !

hairygael, 2018-03-07:
Indeed having ramping is something that has been a work in progress, but
not yet implemented.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

MatsMRL, 2018-03-07:
setAcceleration exists in the Servo service and is implemented at least in Adafruit16CServoDriver, and I think it's also implemented in the Arduino but with a little different algorithm.
Deceleration is a little bit more difficult to implement.
Acceleration = How hard should I step on the accelerator pedal ? Easy.
Deceleration = I press the brake pedal with a specific pressure. At what distance do I need to start braking to come to a full stop at a specific distance? A little more difficult to calculate. 

I think it's also possible to change the speed on the fly using setVelocity.