MyRobotLab/myrobotlab #177: command line -i -install still borked

supertick, 2018-04-07:
it currently does not do anything (although the gui "install all" does)

supertick, 2018-04-10:
worky for me ...

kwatters, 2018-04-10:
On a RasPI,  I download the "latest", did a  "java -jar myrobotlab.jar -install"  .. that completed.
then I run the following command:

java -jar myrobotlab.jar -service runtime Runtime python Python gui SwingGui -invoke python execFile /home/pi/github/pyrobotlab/home/kwatters/Harry.py

The end result is that myrobotlab does not run the script, and oddly enough the SwingGui doesn't launch, it just stays at the agent cli.

kwatters, 2018-07-11:
installing on the raspi seems to work and install everything properly now.  we can close this one.

closed 2018-07-11 by kwatters