MyRobotLab/myrobotlab #194: latest nixie 1.1.71 doesn't download all the dependencies properly

kwatters, 2018-05-11:
I noticed in the most recent build of nixie that not all of the jars are downloaded as expected when services install. 
In particular , the opencv-3.4.1-1.4.1.jar file is missed.    Additionally I see that some of the dl4j dependencies are missed.

I suspect this is an issue with Ivy skipping some jars that it shouldn't, perhaps because of a classifier or something?

supertick, 2018-05-17:
[image]

this is what I got with the latest build....
Is your .m2 damaged?

kwatters, 2018-05-17:
Ok.. so good news..  it turns out I had a stale set of data in my .ivy2 cache.  
Deleteing my  HOME/.ivy2 directory .. and forcing a full reinstall seemed to have resolved the issue with the missing jars.

Closing this issue.