MyRobotLab/myrobotlab #205: Ivy is not processing peer dependencies.

supertick, 2018-05-29:
In Arduino:
meta.addPeer("serial", "Serial", "serial device for this Arduino");

jssc is not being installed

supertick, 2018-05-30:
Hmm .. installing Arduino, does install jssc which is in serial ...
What are you seeing moz4r ?

supertick, 2018-05-30:
[image]

moz4r, 2018-05-30:
Hi ! hehe you put your finger on it : there is a screen that say it is installed BUT there is nothing inside libraries , look inside : 

[image]

supertick, 2018-05-30:
[image]

supertick, 2018-05-30:
Thanks ... working on it now

supertick, 2018-05-30:
Fixed nixie 1.1.172

closed 2018-05-30 by supertick