MyRobotLab/myrobotlab #208: Make (nearly) all Runtime methods & fields non-static

supertick, 2018-06-02:
Runtime is a singleton - there can and always will be only "one" runtime in a running instance of MRL (the other singleton is Security)

many of its methods were made static do to historical evolution into singleton(ish) ..
Now that it is a singleton the preference is to NOT have static methods since "many" Runtime remote serializations could exist in the registry .. if the functions & data are static this will lead to conflict.. 

to do all or any method instead of  static : 
Runtime.methodX()
the preferred alternative is :
runtime = Runtime.getInstance();
runtime.methodX()

kwatters, 2018-07-18:
I'm all for a singleton runtime, however, this will probably break some scripts.
I suspect there's a lot of python scripts that don't differentiate between
"Runtime" and "runtime" because historically both worked...

supertick, 2018-07-18:
Good point ..
And there is certainly value in a single line getting a service ...
e.g.     

`arm = Runtime.get("arm")   #Yay !`

Perhaps a more meaningful refactor would be : 

_all of Runtime's members which "can" be non-static => should be changed  to non-static_ 

as an example : 
 globalArgs is static, and shouldn't be.  If another Runtime from a different instance is serialized into the current running one we cannot show the remote's globalArgs, there is no reason for globalArgs to be a singleton resource.   There **_is_** a reason to have a singleton reference to the local running Security service - no Security service will ever be serialized to another instance (at least I can't think of the use case yet)

-