MyRobotLab/myrobotlab #209: Integrate Tensorflow via JavaCPP as a proper service.

supertick, 2018-06-02:


kwatters, 2018-06-05:
There seem to be some windows dependency issues in the javacpp tensorflow bindings for tesseract 1.7.  I think we should wait until the tensorflow 1.8 bindings are released.  (perhaps we can/should help them with that release.)

kwatters, 2018-06-06:
Turns out Tensorflow has maven java dependencies !  Initial tensorflow service created with dependencies.  no actual implementation yet. but it's borged and ready to be added to.

Of note,  Tensorflow's binaries only support windows/ linux / mac ..  (as far as I can tell, no ARM support at this time.. so if we want to support tensorflow on the raspi or jetson, we'll need to re-visit this implementation.

closed 2018-06-06 by kwatters

saudet, 2018-06-16:
If you guys can come up with linux-arm64 and linux-arm64-gpu builds with cross-compilation support for the JavaCPP Presets for TensorFlow, I'll be happy to maintain them and provide binaries:
https://github.com/bytedeco/javacpp-presets/tree/master/tensorflow