MyRobotLab/myrobotlab #210: MyRobotLabKeyManager

supertick, 2018-06-02:
perhaps its time to start this - where myrobotlab provides cloud services 

* mrl some cloud services could be mrl open source and free (with hit limits)
* mrl cloud service could have shared access to Google/AWS/Azure on a mrl subscription - less cost and higher integration - supply a single key and get access to all big provider functionality integrated with mrl
* mrl access to custom commercial cloud services

moz4r, 2018-06-04:
That is an important talk...
I think it is needed and useful for people want to use extra commercials services with professional support ,and if mrl cannot provide free & open equivalent things. That should remain the soul.
Do you think to borg and expose every services that actually need a key ? 
One key to rules them all !

hairygael, 2018-06-04:
Yes that is a big decision.
It gives a lot more opportunities but changes the spirit of the software
somehow.
It will be hard for some users (myself maybe) that want to be ethically
OpenSource to see the neighbors InMoov be more powerfull because of
commercial services.

Mmmh..

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2018-06-05:
Heh :)
Its actually very inspiring to see such quick reactions :D 

Let's first describe the **intent** so there is no confusion ... from a single word  "MyRobotLabKeyManager" a lot can be guessed, but we all need more information to determine "**intent**"

First and Foremost ...  MRL framework and the services we have created are and always will be Open Source.  In fact, they are Apache License - one of the most unrestrictive open source licenses there is.  This will not change while I am at the helm ...  and if I die while steering the boat, I expect someone will take the wheel and continue a fork with the same license.. (at least that is my hope)

My original intent with this service is to share my API keys.  I have registered for several big company API keys (Google, Microsoft Azure, Amazon, others ....) .  Each of these systems are different and can be complicated to setup and get working.  Some require you to create an account, download an sdk, create subkeys, sub accounts, and follow a series of instructions.  Many of our users are not ready to do this.

My intent was to share my current "big company" subscriptions, so a mrl user with a single key can potentially access all of this "commercial" resources easily.

Why would I do that ?  

Because I'm interested in what creative makers can do, and I want to give them the exposure to BIG COMPANY resources.. without them having to figure out all the crap of a specific company interface.  We did this with Acapela.

Polly is pretty impressive too, and so is Google visual search, Azure AI, and many systems the big companies work on are really very impressive.  I have limited time, and I want our "Open Source" users to tell us the good and bad of the big companies so we might incorporate the good into "our" system.

I know that these big companies have great talent, but they also siphon ideas and work from open source...  **So where does open source siphon good ideas back ?!??!?!?!**  

A real life example follows : 

We found Acapela had a clever idea of adding emotion to voice.  

What happens ? moz4r makes his own emotion system we can add to MarySpeech...  This is when company ideas can **feed back to open source**.

Open Source ?  Remember when you all where using Acapela? .. I refer to an email from me to Gael and Alessandro from 2015

> I recently discovered MaryTTS as an Open Source alternative to Acapela & Google Speech Synthesis - the "default" voice for MaryTTS sounds very primitive - but some of the other voices are on the same level or better than Acapela.  
> MaryTTS has many more controls than the two commercial speech products.  Pitch bending, fine volume control, and audio effects, not to mention its open source so "IT CAN DO ANYTHING !"
> 

I'd be curious to know what you thought my intent was.  I am also curious what you think now.

It "**is**" a very important discussion ...

Respectfully,
Greg