MyRobotLab/myrobotlab #216: ProgramAb HttpClient dependency issue

moz4r, 2018-06-10:
Tested on last build ( without install all, that was the goal of this test, from eclipse it is ok )
seem sraix need it ( seem also we need to use our own services instead of sraix... )
Not sure how to add the right http dependencies version , if I add 

    meta.addDependency("org.apache.httpcomponents", "httpclient", "4.5.2");
    meta.addDependency("org.apache.httpcomponents", "httpcore", "4.4.6");

it crash because : 

java.lang.NoSuchFieldError: java.lang.NoSuchFieldError: DEF_CONTENT_CHARSET

original error :

python.interpreter.3] [ERROR] python error PyException - null Traceback (most recent call last):
  File "<string>", line 20, in <module>
	at org.alicebot.ab.Sraix.pandorabotsRequest(Sraix.java:67)
	at org.alicebot.ab.Sraix.sraixPandorabots(Sraix.java:56)
	at org.alicebot.ab.Sraix.sraix(Sraix.java:44)
	at org.alicebot.ab.AIMLProcessor.sraix(AIMLProcessor.java:419)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1349)
	at org.alicebot.ab.AIMLProcessor.evalTagContent(AIMLProcessor.java:269)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1339)
	at org.alicebot.ab.AIMLProcessor.evalTemplate(AIMLProcessor.java:1451)
	at org.alicebot.ab.AIMLProcessor.srai(AIMLProcessor.java:358)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1345)
	at org.alicebot.ab.AIMLProcessor.evalTagContent(AIMLProcessor.java:269)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1339)
	at org.alicebot.ab.AIMLProcessor.evalTemplate(AIMLProcessor.java:1451)
	at org.alicebot.ab.AIMLProcessor.respond(AIMLProcessor.java:206)
	at org.alicebot.ab.AIMLProcessor.respond(AIMLProcessor.java:179)
	at org.alicebot.ab.Chat.respond(Chat.java:173)
	at org.alicebot.ab.Chat.respond(Chat.java:202)
	at org.alicebot.ab.Chat.multisentenceRespond(Chat.java:225)
	at org.myrobotlab.service.ProgramAB.getResponse(ProgramAB.java:262)
	at org.myrobotlab.service.ProgramAB.getResponse(ProgramAB.java:225)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
java.lang.NoClassDefFoundError: java.lang.NoClassDefFoundError: org/apache/http/client/HttpClient

	at org.python.core.Py.JavaError(Py.java:552)
	at org.python.core.Py.JavaError(Py.java:543)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:190)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:206)
	at org.python.core.PyObject.__call__(PyObject.java:497)
	at org.python.core.PyObject.__call__(PyObject.java:501)
	at org.python.core.PyMethod.__call__(PyMethod.java:141)
	at org.python.pycode._pyx3.f$0(<string>:25)
	at org.python.pycode._pyx3.call_function(<string>)
	at org.python.core.PyTableCode.call(PyTableCode.java:171)
	at org.python.core.PyCode.call(PyCode.java:18)
	at org.python.core.Py.runCode(Py.java:1614)
	at org.python.core.Py.exec(Py.java:1658)
	at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:267)
	at org.myrobotlab.service.Python$PIThread.run(Python.java:159)
Caused by: java.lang.NoClassDefFoundError: org/apache/http/client/HttpClient
	at org.alicebot.ab.Sraix.pandorabotsRequest(Sraix.java:67)
	at org.alicebot.ab.Sraix.sraixPandorabots(Sraix.java:56)
	at org.alicebot.ab.Sraix.sraix(Sraix.java:44)
	at org.alicebot.ab.AIMLProcessor.sraix(AIMLProcessor.java:419)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1349)
	at org.alicebot.ab.AIMLProcessor.evalTagContent(AIMLProcessor.java:269)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1339)
	at org.alicebot.ab.AIMLProcessor.evalTemplate(AIMLProcessor.java:1451)
	at org.alicebot.ab.AIMLProcessor.srai(AIMLProcessor.java:358)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1345)
	at org.alicebot.ab.AIMLProcessor.evalTagContent(AIMLProcessor.java:269)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1339)
	at org.alicebot.ab.AIMLProcessor.evalTemplate(AIMLProcessor.java:1451)
[... 72 lines cut ...]
	at org.alicebot.ab.AIMLProcessor.srai(AIMLProcessor.java:358)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1345)
	at org.alicebot.ab.AIMLProcessor.evalTagContent(AIMLProcessor.java:269)
	at org.alicebot.ab.AIMLProcessor.recursEval(AIMLProcessor.java:1339)
	at org.alicebot.ab.AIMLProcessor.evalTemplate(AIMLProcessor.java:1451)
	at org.alicebot.ab.AIMLProcessor.respond(AIMLProcessor.java:206)
	at org.alicebot.ab.AIMLProcessor.respond(AIMLProcessor.java:179)
	at org.alicebot.ab.Chat.respond(Chat.java:173)
	at org.alicebot.ab.Chat.respond(Chat.java:202)
	at org.alicebot.ab.Chat.multisentenceRespond(Chat.java:225)
	at org.myrobotlab.service.ProgramAB.getResponse(ProgramAB.java:262)
	at org.myrobotlab.service.ProgramAB.getResponse(ProgramAB.java:225)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:188)
	... 12 more
Caused by: java.lang.ClassNotFoundException: org.apache.http.client.HttpClient
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	... 37 more
------

supertick, 2018-06-10:
I'm guessing perhaps ProgramAb has been uber-jar'd or packed with different version of httpclient ?

Although it won't help in this case, there is a lot of services which perhaps could use MRL's HttpClient thereby centralizing the dependency to potentially a single service ...  

Looks like it may be using (and possibly come with) - components 4.2.5
https://github.com/kwatters/program-ab/blob/master/pom.xml

kwatters, 2018-06-11:
is this after an "install all" ? or just trying to install ProgramAB?  We can update ProgramAB to be version 4.5.2 of httpclient.. That's easy enough..

closed 2018-07-11 by moz4r