MyRobotLab/myrobotlab #220: priorities on services shutdown

moz4r, 2018-06-26:
At runtime shutdown, services are calling stopService(). That is great.
I think it will be nice to get priorities on witch service is stopping before an other.
Example, every servo need to be released before arduino service.

moz4r, 2018-06-27:
Hi ! you right I think to take care of attached services is necessary , 
What I want to do ( that is no worky perfect today ) is to rest() whole inmoov skeleton at runtime shutdown. For that I need servocontrol and servocontroller still alive, then they can release.

Solution is to rest() servo inside Servo serviceStop() but I'm not sure everyone want this. but that can be cleaner

supertick, 2018-06-27:
Servo is low level - many things use it
InMoov is high level - it uses many things

Cleanest solution might be for InMoov to register for a shutdown event, so InMoov does the correct thing to shutdown - I would be careful of modifying servo, since many other services depend on it..

e.g. 
subscribe("runtime", "publishShutdown");
InMoov.onShutdown(int seconds) {
            log.info("shutting down InMoov in { } seconds", seconds);
            ... rest all servo services
            ... stop all servo services
            ... stop arduinos
           .... any other cleanup

supertick, 2022-02-06:
all controlled by runtime config
easily changed or modified in a runtime.yml which is processed when you do a release on config

closed 2022-02-06 by supertick