MyRobotLab/myrobotlab #221: Extract frame grabber creation from video processor and put it in opencv.

kwatters, 2018-06-28:
In order to support more flexibility to configure the frame grabber in the OpenCV service, I think we want to pass the frame grabber into the VideoProcessor class.. We should also explicitly allow you to set a frame grabber directly on the OpenCV service and use it if it's set.

supertick, 2018-06-28:
Sounds good
I did a lot of refactoring, but abandoned the changes because I got distracted with something else ...  it certainly could use it...

SQUIRREL !

Spawn32, 2018-06-28:
Thank you guys, you are awesome :)

Spawn32, 2018-06-29:
Hi guy's you both probably already know, opened up a issue over at javacv:

https://github.com/bytedeco/javacv/issues/1021

kwatters, 2018-07-11:
ok. we've exposed the api end point and it would seem there is some success on the TX2.  I have not gotten this approach to work with the raspi, but i think that should be tracked in another ticket.  We should also remember to remove our copy of the OpenCV frame grabber.  Once the next version of javacv is out, we can delete our copy of it, as those changes have been merged upstream. 
closing issue.

closed 2018-07-11 by kwatters