MyRobotLab/myrobotlab #226: MarySpeech - create voices from master list

supertick, 2018-06-30:
add data in loadVoices() with addVoice function from json file here -
https://github.com/marytts/marytts-installer/blob/master/components.json

dependency jars must be added to getMetadata too

LunDev, 2018-06-30:
You might be able to use
[MaryInstaller.getLanguages()](https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/java/org/myrobotlab/maryspeech/tools/install/MaryInstaller.java#L485) & [MaryInstaller.getVoices()](https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/java/org/myrobotlab/maryspeech/tools/install/MaryInstaller.java#L489)
like in [MarySpeech's pre-installation thing](https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/java/org/myrobotlab/service/MarySpeech.java#L126)

> dependency jars must be added to getMetadata too

I don't know if I understood you correctly.
If you refer to the voice-jars: If I remember correctly they were rather large.
(Edit: around a year ago, all voice files were just 5GB in total)

supertick, 2018-06-30:
Ahh .. Thanks MaVo/LunDev :)
I see your installComponentsAcceptLicense methods.
Nice..  are any voices loaded by default ?

LunDev, 2018-06-30:
as of the last time I touched this (way too long ago) no, but I think I read somewhere that @moz4r added one voice (additional to the default one included with mary) as a default to install with the service - not sure

I don't even know if the install-thingy even works anymore after the transition to Maven
(ya know - optional dependecies would be cool? ^^ )

supertick, 2018-06-30:
Ahh .. makes more sense now.  I see @moz4r added that part and started refactoring, I'm now in mid-process of doing a large refactoring.  I see now he's using mrl's dependency downloader in the service's getMetaData() to download a set of voice jars.

   String[] voices = new String[] { "voice-bits1-hsmm", "voice-bits3-hsmm", "voice-cmu-bdl-hsmm", "voice-cmu-nk-hsmm", "voice-cmu-rms-hsmm", "voice-cmu-slt-hsmm",
        "voice-dfki-obadiah-hsmm", "voice-dfki-ot-hsmm", "voice-dfki-pavoque-neutral-hsmm", "voice-dfki-poppy-hsmm", "voice-dfki-prudence-hsmm", "voice-dfki-spike-hsmm",
        "voice-enst-camille-hsmm", "voice-enst-dennys-hsmm", "voice-istc-lucia-hsmm", "voice-upmc-jessica-hsmm", "voice-upmc-pierre-hsmm" };

   for (String voice : voices) {
      meta.addDependency("de.dfki.mary", voice, "5.2");
      if ("voice-bits1-hsmm".equals(voice) || "voice-cmu-slt-hsmm".equals(voice)) {
        meta.exclude("org.slf4j", "slf4j-log4j12");
      }
    }
@moz4r is clever ..

I think I'd still like to present the end user with a "complete" list of all possible voices - and allow them to download more if desired...

kwatters, 2018-06-30:
Those we're all the voices that I found available viaaven , so I added a small loop to include them as dependencies.  

What other voices are available?  Isn't it just sufficient to get the jar of a new voice and make sure it's in the class path?

supertick, 2018-06-30:
heh .. cool .. 
17 are there...
the installer says 34 are available ..

You suggesting get all 34 via Ivy ?

LunDev, 2018-06-30:
@supertick
This will be the right way.
The webgui once *had* pseudo-support for this, but I got lost in trying to show the voice's licenses to the user.
The installer will probably need some love, I don't expect it to work anymore.

@kwatters 
I haven't researched what the maven packages supply, but some voices wanted additional files.

supertick, 2018-06-30:
This is the data I see coming out of the installer ..
Some voices look to be the same "voice" but different implementations e.g. hidden markov, vs some other flavor ..

{bits1=bits1,
bits1-hsmm=bits1-hsmm,
bits2=bits2,
bits3=bits3,
bits3-hsmm=bits3-hsmm,
bits4=bits4,
cmu-bdl=cmu-bdl,
cmu-bdl-hsmm=cmu-bdl-hsmm,
cmu-nk-hsmm=cmu-nk-hsmm,
cmu-rms=cmu-rms,
cmu-rms-hsmm=cmu-rms-hsmm,
cmu-slt=cmu-slt,
dfki-obadiah=dfki-obadiah,
dfki-obadiah-hsmm=dfki-obadiah-hsmm,
dfki-ot=dfki-ot,
dfki-ot-hsmm=dfki-ot-hsmm,
dfki-pavoque-neutral=dfki-pavoque-neutral,
dfki-pavoque-neutral-hsmm=dfki-pavoque-neutral-hsmm,
dfki-pavoque-styles=dfki-pavoque-styles,
dfki-poppy=dfki-poppy,
dfki-poppy-hsmm=dfki-poppy-hsmm,
dfki-prudence=dfki-prudence,
dfki-prudence-hsmm=dfki-prudence-hsmm,
dfki-spike=dfki-spike,
dfki-spike-hsmm=dfki-spike-hsmm,
enst-camille=enst-camille,
enst-camille-hsmm=enst-camille-hsmm,
enst-dennys-hsmm=enst-dennys-hsmm,
istc-lucia-hsmm=istc-lucia-hsmm,
marylux=marylux,
upmc-jessica=upmc-jessica,
upmc-jessica-hsmm=upmc-jessica-hsmm,
upmc-pierre=upmc-pierre,
upmc-pierre-hsmm=upmc-pierre-hsmm}

supertick, 2018-06-30:
on closer inspection - these potentially are all different voices, its just a little hard to tell with from the keys they are identified with, but the VoiceComponentDescription coming back from the installer has all the details.  @kwatters so what do you recommend ?  install all 34 with ivy ?  If not there should be an easy way to get more if desired.

kwatters, 2018-06-30:
Install via ivy/maven.  It's easy, clean, and maintainable.  You want a new voice add it to the dependency list.  No confusion about what has or hasn't been installed.

supertick, 2018-07-01:
alright - the caching "is" pretty nice with maven/ivy.
I'll add the rest, the installer will be certainly useful for finding additional - a perhaps I might re-visit installed vs available later .. the licensing will need to be addressed at some point too (in a generalized manner in the AbstractSpeechSynthesizer or even more generally "ie - Service")

supertick, 2018-07-01:
@kwatter - ok, now it makes sense what you are saying, only 17 were uploaded into bintray, so only 17 are available through a maven repo - I've done a little experimentation and as far as I know the others cannot be found anywhere.  The nginx webserver marytts group maintains does not allow browsing and all preliminary references come back with a 404.

I've asked the MaryTTS group what their future plans are, I agree maven repo distribution would be preferable.   

If they plan to go that way the mary installer will become obsolete.

supertick, 2018-07-01:
Ahahah ..
The MaryTTS project is great .. it is funny how they have evolved so similarly to MRL.  They started with a home-brew installer and resource loader with a custom repo on github, then they moved to gradle and Ivy ! :D 

Here is an example of running their installer with a missing bits2 voice

BUILD SUCCESSFUL in 1m 24s
3 actionable tasks: 2 executed, 1 up-to-date
C:\github\marytts-installer>marytts install voice-bits2
Download http://mary.dfki.de/repo/de/dfki/mary/voice-bits2/5.2/ivy-5.2.xml
Download http://mary.dfki.de/repo/de/dfki/mary/voice-bits2/5.2/voice-bits2-5.2-data.zip
Download http://mary.dfki.de/repo/de/dfki/mary/voice-bits2/5.2/voice-bits2-5.2.jar
> Task :verify
Verify  voice-bits2-5.2-data.zip
Verify  voice-bits2-5.2.jar

> Task :install
Unpack  halfphoneFeatures_ac.mry
Unpack  halfphoneUnits.mry
Unpack  joinCostFeatures.mry
Unpack  timeline_basenames.mry
Unpack  timeline_waveforms.mry
Unpack  de_english-words.txt
Unpack  de_unknown.txt
Copy    voice-bits2-5.2.jar
Copy    marytts-lang-de-5.2.jar
BUILD SUCCESSFUL

supertick, 2018-07-26:
MarySpeech seems to work in the new AbstractSpeechSynthesis context, which was a pretty big refactor.  

The guidelines now of creating a new speech synthesis are pretty easy implement, you just need to use the addVoice function to add a voice and generateAudioData to generate the audio data.

The abstract handles all the pub/sub paths, UI, filtering of input, overlay of effects sound files, and caching of audio data.  

I think the result came out pretty well, with possibilities of handling SSML in the future ...

moving to Test column

supertick, 2018-07-31:
fixed - "all" speech synthesis services now handle processing in almost identical ways.  Additionally, the licensing/dependency loading should be handled at the getMetaData level (Mary's is now handled that way - licensing / prompting should be a feature we add)

closed 2018-07-31 by supertick