MyRobotLab/myrobotlab #229: Fix dependencies so that InMoov2 does not reuse anything from InMoov normal dependencies.

kwatters, 2018-07-02:
When developing InMoov2  you must not break anything in InMoov.

This means you must define a NEW dependency for inmoov2.

Do NOT change InMoov.  If you want to experiment with InMoov2. 

As it stands now, there are multiple versions of the inmoov.fr.inoov dependency defined in the POM.

MULTIPLE VERSIONS OF THE SAME DEPENDENCY DO NOT WORK!  YOU MUST DEFINE ONE AND ONLY ONE VERSION FOR ALL DEPENDENCIES IN THE POM.

Create a new repository for "inmoov2"   or create a branch of the InMoov repostiory and publish a different artifact from that branch. 

The artifact MUST HAVE A NEW GROUP/ARTIFACT ID!

call it   "inmoov.fr.inmoov2"    DO NO REUSE "inmoov.fr.inmoov"

hairygael, 2018-07-02:
Thanks Kevin,
We also discussed with Greg to update the i01 to i02.
The develop branch has been recently changed and makes the use of inmoov
instead of i02 in all the gestures.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-07-02:
i01 is no longer required to evolve and is fixed on version 1.0.0 of the dependency, which interests us here it is around the directory gestures and chatbot. I think it makes more sense to maintain the same base rather than to multiply things. At the end i02 should be renamed i01 .. 
Ohh we should doing it NOW maybe !

kwatters, 2018-07-02:
Do not change i01 in the process of creating i02.  You should treat i02 as a completely new service and you should not be changing any functionality of the current i01 service.

I think your work should move to another branch and when it's ready make a pull request so we can review and merge it in.

moz4r, 2018-07-02:
i01 is not changed, I understood one of the interests of maven migration is the abilitie to drive multiple versions of dependencies. i01 is using v1.0.0 . may be mistaken.

The goal is inmoov repo contain only few necessaries stuff like gesture and aiml files. that is the current WIP : conceal the existing and evolve many python files that have no place in the repo.

The question is do we want to maintain 2 inmoov service. If yes so we need a new repo maybe. personally I vote to merge asap and focus to get inmoov2 worky. close goal. But we can dedicate a i02 dependencies also.. I don't like it :)

kwatters, 2018-07-02:
I think you are fundamentally misunderstanding something.

When defining a maven dependency it is uniquely defined by the combination of the group and artifact id.

You can only include 1 version of any dependency in maven project.

This means, when you defined version 1.0.1 of InMoov.fr.inmoov that you it will not install 1.0.0.  

I think fundamentally InMoov2 will be different from inmoov1.  It should be a completely new service because some people will still want to use inmoov1.  

For that reason you must not break inmoov1 while working on the new one.

supertick, 2018-07-02:
Kevin is right about dependency management.
Only one dependency (group-id artifact-id) will be installed, this is the point of having a dependency management system.

My impression was InMoov2 was created because it was going to be a big fork of changes - for example of how it is configured.  
If that was the intent and the dependency needs change to support these new InMoov2 changes, then the artifact-id should probably change to inmoov2 
That would be cleanly forking the service and its dependency which would allow both to operate.

moz4r, 2018-07-02:
ok let's (full) fork ... Inmoov repo will contain Inmoov and Inmoov2 soon. With 2 build.xml ?
Inmoov folder will contain previous master branch things.

kwatters, 2018-07-02:
you could just create a branch...