MyRobotLab/myrobotlab #232: Telephone Service

supertick, 2018-07-03:
Telephone service which allows bot to dial.  This was inspired from @MikeG complaining his inmoov doesn't dial someone when it says it will.

References :

- https://dzone.com/articles/jtapi-hands-part-i
- http://gusclass.com/blog/2014/01/14/using-the-hangouts-telephone-api/

hairygael, 2018-07-04:
Nice!
In the AIML there is a function to launch a phone call, send emails, but it is out of order and doesn't work.
In fact when we saw each other in Paris (Greg), I mentioned a very annoying bug regarding "contacts".
So first the related SRAIX is not doing it's job (broken link at Pandora) but then it redirects on another AIML file which will read slashs and dots instead of contact names.

So if we could find another option than Pandora, it would be great!

moz4r, 2018-07-09:
oooh we have : https://github.com/MyRobotLab/inmoov/blob/master/InMoov/gestures/phonehome.py :)

hairygael, 2018-07-09:
:)
With mp3 sound!!
[](https://github.com/MyRobotLab/inmoov/blob/master/InMoov/system/sounds/E%2CT%20phone%20the%20big%20home%20of%20the%20inmoov%20nation.mp3)
But more serious we have:
[](https://github.com/MyRobotLab/inmoov/blob/master/InMoov/chatbot/bots/en/aiml/contactfunctions.aiml)