MyRobotLab/myrobotlab #245: webgui fails to start in latest

kwatters, 2018-07-26:
On harry, i'm seeing the following exception in gson on startup.  basically this happens when the webgui launches and we open the browser to the i01.ear service page directly.  
i've deleted the .myrobotlab .. this is a fresh install of the "latest" as of today.   I didn't have this issue a few weeks ago.. though I'm not sure which build the issue first showed up .

20:31:50.937 [New I/O worker #8] WARN o.a.w.DefaultWebSocketProcessor [DefaultWebSocketProcessor.java:569] Failed invoking AtmosphereFramework.doCometSupport()
java.lang.StackOverflowError: null
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:546) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:553) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:546) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:553) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:546) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:553) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:546) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:553) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:546) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:553) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types.canonicalize($Gson$Types.java:111) ~[myrobotlab.jar:1.1.234]
        at com.google.gson.internal.$Gson$Types$WildcardTypeImpl.<init>($Gson$Types.java:546) ~[myrobotlab.jar:1.1.234]

kwatters, 2018-07-27:
Which unit test ?    SpeechSynthesisTest.java ?

supertick, 2018-07-27:
I think this is good, exploring what exactly is being tested .. and perhaps a discussion of the Test service

supertick, 2018-07-28:
Hmm.. can't replicate ... starting with START_INMOOV.bat everything appears to be worky, virtual inmoov comes up - can talk with it ... it calls me "Grant" vs "Greg"  :)

supertick, 2018-07-29:
need help to replicate

kwatters, 2018-07-31:
this is/was related to the embeddedSolr service not being marked as transient.  This would have been a difficult case to have auto-testing elves to catch because normally this object is null (by default) so unless you automated a test to start the embedded solr server and then tested serialization, we wouldn't have caught it.  

I pushed a fix for this specific issue and I'll work towards pulling something similar to the Test.serialization test method into a unit test.. we might catch some stuff.. but definitely not everything.

kwatters, 2018-08-14:
I pushed a unit test for testing of serialization.. closing this issue as we don't see it anymore.

closed 2018-08-14 by kwatters