MyRobotLab/myrobotlab #256: Virtual inmmov seem borked

moz4r, 2018-09-10:
tested from 2 computers, not sure if it is me or not.
test with full and also service test script.
JME windows stay black and stucked, that crash mrl.

moz4r, 2018-09-10:
At first glance, worky inside eclipse, noworky from standalone jar

moz4r, 2018-09-11:
dependencies issue ! need to find it ...

moz4r, 2018-09-11:
ok that was a very strange bug. race condition about jme3 audio engine and maryTts. Maybe not the only thing but worky now like before.

hairygael, 2019-04-13:
testing version 1.1.157 (although the servos cannot be testing proprerly because of autoEnable refactor)
I realize Virtual InMoov is borked.
It just doesn't move and stands in a frozen position. This must be related to a lot of refactoring with virtual stuff.

supertick, 2021-08-08:
[image]
If started from the InMoov2 UI - it appears to work..
JMonkeyEngine will also start by just running
jme = runtime.start('jme','JMonkeyEngine')
But in this case it will not load all InMoov models etc - as this is not inherently JMonkeyEngine's design.
It can be done manually with more code if desired though.

Setting this to verify - will close if no activity.

hairygael, 2021-08-09:
I just verified and it is okay and worky with version 534

closed 2021-08-11 by supertick