MyRobotLab/myrobotlab #258: Format chatbot UI

moz4r, 2018-09-12:
just to make things a little clearer and tidy

[image]

supertick, 2018-09-12:
I think it should follow a regular chatbox format ...
e.g.
**Anthony** :  Hello bot
**en-US** : Hello Anthony, its great seeing you.
**Anthony** : What is going on ?
**en-US** :  The weather is 30 degrees Celsius and slight winds from the northeast.

LunDev, 2018-09-12:
either that or the aligned version: e.g. align the bot's message to the left and the user's to the right
@supertick 's version is easier to implement, so I vote for that one :p

moz4r, 2018-09-12:
great ! we are agree the screenshot is what we have today

moz4r, 2018-09-17:
more sexy, progress

[image]

hairygael, 2018-09-17:
I agree as well, Grog's version is tidy.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2018-10-12 by moz4r