MyRobotLab/myrobotlab #263: HttpClient versions - stop the madness

supertick, 2018-09-15:
Currently Runtime meta data is defined as :

meta.addDependency("org.apache.httpcomponents", "httpclient", "4.5.2");

ProgramAB is
    meta.addDependency("org.apache.httpcomponents", "httpclient", "4.5.2");
    meta.addDependency("org.apache.httpcomponents", "httpcore", "4.4.6");

If httpclient is critical to the very core of mrl - then perhaps all version references should be removed - and the Runtime get a "full" description

    meta.addDependency("org.apache.httpcomponents", "httpclient", "4.5.2");
    meta.addDependency("org.apache.httpcomponents", "httpcore", "4.4.6");

Anyone disagree ?

wilcovantoorn, 2018-09-15:
It Would be the most logical approach.

kwatters, 2018-09-25:
who ever gets around to it, http client/core should be excluded from all services and included in either runtime or in httpclient.  (my preference would be runtime.. for simplicity)

closed 2018-10-07 by supertick