MyRobotLab/myrobotlab #264: Low level Services should be unaware of high level Services

supertick, 2018-09-17:


supertick, 2018-09-17:
Some care and consideration should be taken when modifying services which share functionality to other services.  

Arduino, Servo, SwingGui, WebGui are "low level".
InMoov, Tracking, WorkE are "high level" services which use low level services.

What problem is the following code solving ?
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/java/org/myrobotlab/service/Arduino.java#L1888-L1897

perhaps we could add the functionality, but minimally - we need to add it in a generalized way.

supertick, 2018-09-17:
I got this ...
moz4r explained that InMoov before shutdown should set all servos to rest position..

I plan to make the following method in Service
void makeReadyForShutdown()

The default in Service will do nothing, however, a service like InMoov can override it and do everything it needs to before shutting down - for example make all servos rest, and say its going to shutdown .  

Initially, I'll keep it simple and this will be driven by Runtime - you can block, but in the future Runtime might spawn a thread if the service in question is not behaving, and come back with a hammer...

moz4r, 2018-09-18:
I like it, I wonder how to be sure arduino is not released before I send instructions to servo ?

supertick, 2018-09-18:
https://github.com/MyRobotLab/myrobotlab/pull/266

supertick, 2018-09-20:
fixed

moz4r, 2018-09-26:
I have some cleanup to to inside inmoov.stopservice()

MatsMRL, 2018-09-27:
The code that you are refering to is to ensure that higher level services can make some cleanup before being shutdown. For example, if I have a motor runnig, I want to be able to sen a stop command to it, before the Arduino / Serial service stops.

supertick, 2018-09-27:
Right .. now when Runtime is requested to shutdown, it will call makeReadyForShutdown on each running service.  This will give the opportunity for each service currently running to do what it has to, to safely shutdown (including stopping motors).  

The only additional change I have not pushed to the develop branch is renaming it preShutdown as per suggestion...   I'd figure I'd wait for moz4r to push his changes in first.

MatsMRL, 2018-09-27:
OK. So how does this work if the Ardino service is shutdown ? How does the Motor service get informed about that event ?

supertick, 2018-09-27:
The assumption is InMoov wants to shutdown ...
The appropriate way to shutdown everything is to call Runtime.shutdown() 
When this is called, all services have an opportunity to override there makeReadyForShutdown () (soon to be preShutdown()) and do whatever cleanup is necessary...

In InMoov's case .. InMoov would override .  e.g.
`
public void makeReadyForShutdown () {
      //  put servos back to rest            
      rest();
     // wait a couple seconds to get into rest position
      sleep(2);
      // detach gracefully
      detach();
      // do anything more desired
      speak(" goodbye cruel world ");
      ...
`
Since InMoov service is the highest level it should do the orchestration to guarantee a pleasant shutdown.  Motor and Arduino are lower level services and do not need to worry or do any additional activities ... I think ..

closed 2018-10-07 by supertick