MyRobotLab/myrobotlab #272: runtime.restart() no worky

moz4r, 2018-10-01:
java.io.FileNotFoundException: msgs\agent.1538425273072.part (Le chemin dâaccÃ¨s spÃ©cifiÃ© est introuvable)
        at java.io.FileOutputStream.open0(Native Method)
        at java.io.FileOutputStream.open(Unknown Source)
        at java.io.FileOutputStream.<init>(Unknown Source)
        at java.io.FileOutputStream.<init>(Unknown Source)
        at org.myrobotlab.io.FileIO.toFile(FileIO.java:1301)
        at org.myrobotlab.io.FileIO.toFile(FileIO.java:1320)
        at org.myrobotlab.service.Runtime.restart(Runtime.java:1975)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:188)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:206)
        at org.python.core.PyObject.__call__(PyObject.java:480)
        at org.python.core.PyObject.__call__(PyObject.java:484)
        at org.python.core.PyMethod.__call__(PyMethod.java:126)
        at org.python.pycode._pyx36.f$0(<string>:1)
        at org.python.pycode._pyx36.call_function(<string>)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyCode.call(PyCode.java:18)
        at org.python.core.Py.runCode(Py.java:1614)
        at org.python.core.Py.exec(Py.java:1658)
        at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:267)
        at org.myrobotlab.service.Python$PIThread.run(Python.java:159)

supertick, 2019-01-03:
Moving to Ogre.  Changed UI button to "Exit"
We still want this restart, but it needs to be done by a "thin Agent client".

Currently the Agent uses std:in to send info to the mrl instance it controls, but does not get info from the mrl instance.  std:out is not an option, because "everything else" dumps info to std:out.  What is needed is a network connection.

The Agent in Ogre release should connect to an MRL instance with socket or websocket connection - and use standard messaging over that connection.  Then the MRL instance can tell the Agent .. "restart me" and the Agent will do it.

supertick, 2022-02-06:
totally worky

closed 2022-02-06 by supertick