MyRobotLab/myrobotlab #276: Massive Reformat to standardize all Java files

supertick, 2018-10-01:
Make sure all java files follow the conventions we have currently checked in the root of the project (mrl_java_formmatter.xml) in a single commit.

LunDev, 2018-10-01:
I would vote in favor of [EditorConfig](https://editorconfig.org/) to support other IDEs as well. Maybe?

kwatters, 2018-12-03:
My request is that we make one small change to the formatting before we do the full cleanup.

I'd like to increase "org.eclipse.jdt.core.formatter.comment.line_length" value="80"
to be "132"    

When auto formatting, long comments tend to do odd things.

kwatters, 2018-12-31:
I also think we should standardize on unix style line endings..  I'm just not 100% certain the best way to enforce that.

LunDev, 2018-12-31:
Git has some very nice configuration options:
```
core.eol
core.autocrlf
```
Those could be set locally (git config) and possibly even global (git config --global). (Just be careful when changing already cloned repos).
But they could also be set for a repository (and thus enforcing it für everyone - even those lazy ones who don't adjust their local config - no, I'm definately not talking about myself here) (.gitattributes file).

For new files and all local changes, [EditorConfig](https://editorconfig.org) is quite a nice tool. This snippet (copied from their docs), will ensure that every program that supports EditorConfig will create new files with lf line endings:
```
# EditorConfig is awesome: https://EditorConfig.org

# top-most EditorConfig file
root = true

# Unix-style newlines with a newline ending every file
[*]
end_of_line = lf
insert_final_newline = true
```
With GitHub, Eclipse, IntelliJ, NetBeans, VisualStudioCode, Notepad++, Brackets, Atom, Sublime, gedit, vim, emacs and many more everything should be supported.

I know of two possible ways to prevent commits with crlf from ending up upstream:
1. using a local pre-commit hook to check and reject such commits
2. checking this in CI (and fail the build and hopefully prevent it from getting merged)
2b. adding a post-receive hook to check and do the same
2c. an GitHub app for this purpose
2d. maybe GitHub Actions could work for this?

Don't know what would be best for MyRobotLab, these are just the options I know of (and currently use or have used in the past).

kwatters, 2019-01-03:
I reverted the change to comment line length to be 80.  this has been merged to develop.

closed 2019-01-03 by kwatters