MyRobotLab/myrobotlab #278: OpenCV Exceptions in VP do not stop capture

supertick, 2018-10-02:
currently :
an endless stream of exceptions are logged when an exception occurs in the VideoProcessor and does not successfully stop the capture - this made huge logs in junit tests

moz4r, 2018-10-03:
oh, related ? https://github.com/MyRobotLab/myrobotlab/issues/284

moz4r, 2018-10-09:
I still have issues with VideoInputFrameGrabber : 

the first 

opencv.capture()
opencv.stopCapture()

seem capturing..

the second crash : 

```
org.bytedeco.javacv.FrameGrabber$Exception: videoInput.setupDevice() Error: Could not setup device.
	at org.bytedeco.javacv.VideoInputFrameGrabber.start(VideoInputFrameGrabber.java:111)
	at org.bytedeco.javacv.VideoInputFrameGrabber.start(VideoInputFrameGrabber.java:101)
	at org.myrobotlab.opencv.VideoProcessor.run(VideoProcessor.java:275)
	at java.lang.Thread.run(Unknown Source)
------

01:01:57.111 [opencv_videoProcessor] INFO  c.myrobotlab.opencv.VideoProcessor - beginning capture
01:01:57.123 [opencv_videoProcessor] ERROR c.myrobotlab.opencv.VideoProcessor - stopping capture
org.bytedeco.javacv.FrameGrabber$Exception: videoInput is null. (Has start() been called?)
	at org.bytedeco.javacv.VideoInputFrameGrabber.grab(VideoInputFrameGrabber.java:162) ~[javacv-1.4.2.jar:1.4.2]
	at org.myrobotlab.opencv.VideoProcessor.run(VideoProcessor.java:292) ~[classes/:na]
	at java.lang.Thread.run(Unknown Source) [na:1.8.0_181]
```

supertick, 2018-11-24:
should be fully worky / restartable..

closed 2018-11-24 by supertick