MyRobotLab/myrobotlab #284: OpenCv UI issues

moz4r, 2018-10-03:
Just found one, on the default screen we see after service started, videoinput selected, after clicked to capture, no "stop capture" button

kwatters, 2018-10-03:
yup. i'm pretty sure grog fixed this.. i also had the same fix on the lloyd2 branch.

hairygael, 2018-10-07:
When launching:
 **opencv.capture()** or **opencv.stopCapture()**
It automaticaly resize the SwinGui to a disproportianal size regarding the screen.

moz4r, 2018-10-09:
reproduced, oh also opencv.setFrameGrabberType() do not broadcastState(), so a different framegrabber is showed inside UI

supertick, 2018-11-24:
should all be fixed ... 
capturing is a lot more code friendly ..
cv.capture("someFile.mp4") 
cv.capture("http://somewebpicture.jpg") 
cv.capture("https://www.youtube.com/watch?v=SLp5YVplJAQ") 

looked for cached or recorded files in the OpenCV directory

closed 2018-11-24 by supertick