MyRobotLab/myrobotlab #289: coco.names translation

moz4r, 2018-10-04:
If it is enough to translate coco.names, maybe we can add some localised coco.names. based on Locale format ? And add a language parameter before filter start ?

kwatters, 2018-10-04:
Yes, this file contains all of the "labels".  We can translate these to any language and load them.  One thing to note,  I don't think that cvPutText support any UTF-8 characters, so that might be a challenge to display it.

Also, there are places in the code that I'm doing things that rely on the english version of the label.  like  yoloDetectedObject.label.equals("person")  ..  so I think we need a way to preserve some normalized behavior like this.

moz4r, 2018-10-05:
intersting, so we need to keep orignal english names for post processing. and think of maybe of a localized publisher or other thing.
For now will try to override modelNames from python land

moz4r, 2018-11-06:
We can do it better, but worky for now. 
coco.names updated from python land if user set a custom language pack

closed 2018-11-06 by moz4r