MyRobotLab/myrobotlab #291: ConcurrentModificationException

moz4r, 2018-10-05:
To not forget, I have sometime this random error

java.util.ConcurrentModificationException
	at java.util.TreeMap$PrivateEntryIterator.nextEntry(Unknown Source)
	at java.util.TreeMap$EntryIterator.next(Unknown Source)
	at java.util.TreeMap$EntryIterator.next(Unknown Source)
	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:206)
	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.write(ReflectiveTypeAdapterFactory.java:125)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.write(ReflectiveTypeAdapterFactory.java:243)
	at com.google.gson.Gson.toJson(Gson.java:669)
	at com.google.gson.Gson.toJson(Gson.java:648)
	at com.google.gson.Gson.toJson(Gson.java:603)
	at com.google.gson.Gson.toJson(Gson.java:583)
	at org.myrobotlab.codec.CodecUtils.toJson(CodecUtils.java:392)
	at org.myrobotlab.framework.Service.save(Service.java:1743)
	at org.myrobotlab.service.abstracts.AbstractSpeechSynthesis.getVoices(AbstractSpeechSynthesis.java:673)
	at org.myrobotlab.swing.abstracts.AbstractSpeechSynthesisGui$3.run(AbstractSpeechSynthesisGui.java:331)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

supertick, 2018-10-07:
How to reproduce ?  When/Where is the error ?

supertick, 2018-10-09:
was it a bad json file ?
If you remove the json and start "fresh" - does it happen again ?

moz4r, 2018-10-11:
fount it, race condition if using save() and do a voiceList operation at same time. we don't need to writeOut voiceslist immediatly, will disable it.

closed 2018-10-12 by moz4r