MyRobotLab/myrobotlab #294: opencv.removeFilters(), sometime, noworky...

moz4r, 2018-10-05:
no more informations for now..
no errors, filters still loaded

moz4r, 2018-10-05:
i01.opencv.capture()
i01.opencv.removeFilters()
i01.opencv.addFilter("PyramidDown")
i01.opencv.addFilter("Yolo")
sleep(5)
i01.opencv.stopCapture()
i01.opencv.removeFilters()

moz4r, 2018-10-05:
after shutdown : 

21:41:12.967 [i01.opencv] ERROR c.m.f.Service [Service.java:2073] i01.opencv error InvocationTargetException - null java.lang.reflect.InvocationTargetException
	at sun.reflect.GeneratedConstructorAccessor67.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown Source)
	at java.lang.reflect.Constructor.newInstance(Unknown Source)
	at org.myrobotlab.service.OpenCV.createFrameGrabber(OpenCV.java:491)
	at org.myrobotlab.service.OpenCV.capture(OpenCV.java:435)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.myrobotlab.framework.Service.invokeOn(Service.java:1370)
	at org.myrobotlab.framework.Service.invoke(Service.java:1306)
	at org.myrobotlab.framework.Service.run(Service.java:1706)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.OutOfMemoryError: Physical memory usage is too high: physicalBytes (7265M) > maxPhysicalBytes (7232M)
	at org.bytedeco.javacpp.Pointer.deallocator(Pointer.java:588)
	at org.bytedeco.javacpp.Pointer.init(Pointer.java:124)
	at org.bytedeco.javacpp.opencv_core$Mat.allocate(Native Method)
	at org.bytedeco.javacpp.opencv_core$Mat.<init>(opencv_core.java:19261)
	at org.bytedeco.javacv.OpenCVFrameGrabber.<init>(OpenCVFrameGrabber.java:91)
	... 13 more

moz4r, 2018-10-05:
seem yolo thread still running, after filter removed

moz4r, 2018-10-09:
opencv.removeFilters() is ok now.
@kwatters if you can find a solution for memory that increase at every launch of filter, it is very welcome !

hairygael, 2018-10-26:
Hello guys,
I just tried version 1.1.952 with launching openCV. The swinGui window over expended my screen just like it use to do before. :(
Then I tried "face detection" voice command, openCV UI freezed. I have sent a noworky just in case.
:)

moz4r, 2018-10-28:
Hi Gael, I ear hammer from @GroG forge about openCV polish, a little mishap, someone stole his anvil with code inside

moz4r, 2018-11-05:
Fresh test on the robot (4gb ram) I can launch only twice the filter, until it overflow memory and crash the system.

moz4r, 2018-11-21:
1.1.14 testing : 

Wahoo ! no more RAM issue, but this cause runtime random crash, after filter released

[hs_err_pid8684.log](https://github.com/MyRobotLab/myrobotlab/files/2605156/hs_err_pid8684.log)

supertick, 2018-11-24:
you can use filter.enable()/filter.disable()
there can still be issues caused by invalid states of filters as the pipeline changes.
Its cleaner and less problematic if you load all filters your interested in the pipeline and just enable/disable them at the appropriate time.

closed 2018-11-24 by supertick