MyRobotLab/myrobotlab #297: OOB tag appears in Chatbot  tab

hairygael, 2018-10-06:
[image]

LunDev, 2018-10-06:
I think that's the result of @moz4r 's recent ProgramAB UI overhaul.
-> https://github.com/MyRobotLab/myrobotlab/pull/265/files#diff-550e151368a81f46457188f61c186271R222

hairygael, 2018-10-06:
Yes it seems line 63 shouldn't have spaces
name: " > oob <"
name: " >oob<",

LunDev, 2018-10-06:
You know that the file/line you're referencing is used in the WebGui, right?
I had linked to L222 of ProgramABGui.java (Swing Gui), `&gt;` is acutally the reason for the `>`.
Besides that, I think this change was actually intended by @moz4r , so he might have to elaborate further.

hairygael, 2018-10-06:
Hey no! I was just looking around for something with oob and space.
And actually didn't even see the line 222.
Now  OOB: </b>
in the AIML, I had to remove sometime ago some </br> that would also cause to appear in the chatbot tab.
<template>Daisy, Daisy. Give me your answer do.<br/>I am half crazy. 
		All for the love of you.<br/>It won't be a stylish marriage. 
		I can't afford a carriage.<br/>But you'll look sweet. Upon the seat.<br/>Of a bicycle built for two.</template>
To fix it I had to do:
<template>
            Daisy, Daisy. 
            Give me your answer do. 
            I am half crazy. 
            All for the love of you. 
            It won't be a stylish marriage. 
            I can't afford a carriage. 
            But you'll look sweet. 
            Upon the seat. 
            Of a bicycle built for two.
         </template>

LunDev, 2018-10-06:
You might want to ask @moz4r for help, I don't understand your comment completly.

hairygael, 2018-10-06:
Yes @moz4r will see that.
Thanks MaVo

moz4r, 2018-10-08:
the <br/> will not cause issue anymore ( Inside UI rendering ), now filtered
Concerning OOB tag appears, useful to know what the robot "do" behind our back

kwatters, 2018-10-11:
@moz4r  I think this issue is a side effect of the UI stripping html.  The UI should not do that in my opinion (as i stated in the code review.)  Perhaps, that should be reverted?

moz4r, 2018-10-11:
Maybe, now we have UI that interpret html. 
mmm. We just need to test rendering and ajust things like double spaces and cariage return with nothing after.

kwatters, 2018-10-11:
why not just let the html pass through as it did before?

moz4r, 2018-10-11:
sometime, result is very ugly. we need to tweak a little..

kwatters, 2018-10-11:
But those ugly results should be fixed in the AIML , no? I would think that's better than removing the ability to have HTML in the response..

moz4r, 2018-10-22:
OOB debug moved to debug widget.
<br/> tag is now interpreted Inside UI

closed 2018-10-22 by moz4r

hairygael, 2018-10-23:
> OOB debug moved to debug widget.
>
> tag is now interpreted Inside UI

Handy and good!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>