MyRobotLab/myrobotlab #302: Implement / check isReady() for speech services and broadcast status

moz4r, 2018-10-08:


supertick, 2018-10-09:
what would be "ready" - keys ? internet connection ? other ?

kwatters, 2018-10-09:
we have "ready" on ProgramAB currently to let us know if it has or hasn't loaded it's brain yet.  we should consider this as part of the basic lifecycle of a service to let us know that we've asked it to start loading some data.. and when it's ready, we should have a callback to indicate it's ready.  All services should probably implement this.   
especially services that load large resources  (like dl4j)

supertick, 2018-10-09:
http://myrobotlab.org/content/service-life-cycle

Service.isReady() has always been part of the interface.  It's left up to an implementation of a service to override.   

locally I've added a protected isReady to Service.

The method is defaulted to be true and now returns the protected member.  The reason to change it to a protected member is for the json representation the service uses to describe state.  This is useful, since webgui & external systems can look at the field in json, but are not able to call the method isReady().

I'll push my small changes momentarily, but its up to your specific service to check & notify if not ready..

supertick, 2018-10-10:
This "could" be put into AbstractSpeechSynthesis, but I don't see the point as missing keys are already reported back to the Gui without it.
```java
 public boolean isReady() {
    String[] kn = getKeyNames();
    if (kn != null) {
      for (String keyName : kn) {
        String key = getKey(keyName);
        if (key == null) {
          warn("required key %s not set", keyName);
          isReady = false;
          return isReady;
        }
      }
    }
    isReady = true;
    return isReady;
  }
```
The service really isReady for user input, its just the user input that is missing.
I think perhaps my intention originally for this method was for services which take a while to load vs require user input.  For example the Arduino serial connection does take some millis to start, and if attaching servos happens immediately after - it could be a problem for the user. 

Funny that this was originally for Arduino, but appears to have been stripped out.

The facility is there, but its up to the individual implementation to make this useful.  Going to close unless there is some objection.

moz4r, 2018-10-10:
you right, hijacked :) 
I want to use it also for MarySpeech, can take long time to load on small configurations.
will change few code lines inside speech services : 

- add an icon red/green
- add service class name, for information
- add setIsReady(bool) Inside service, to broadcast state

oh, do we have a class to check internet connexion ?

kwatters, 2018-10-10:
This should be abstract and it should be implemented by the subclass.

closed 2018-10-12 by moz4r