MyRobotLab/myrobotlab #306: SwinGui freeze or very slow to react

hairygael, 2018-10-09:
SwinGui freeze or very slow to react after a few vocal commands.

Last vocal used before it acted strangely was "random life"-->VieAleatoire.py
I cannot send a noworky because the window doesn't react.
I need to force shutdown via task manager.

Also the command "face detection" freezes the opencv window, and freezes the whole swingui.
I need to force shutdown via task manager.

Also the command "camera on" doesn't start the capture in the opencv window, and freezes the whole swingui. Oddly the picture taken during the startup sequence of InMoov is upside down.
I need to force shutdown via task manager.

moz4r, 2018-10-09:
will check, I also noticed some issues, related to filters and unidentified (yet) memory overflow.
Also we need to work on this, will PR some mods soon to talk about.

https://github.com/MyRobotLab/myrobotlab/issues/295
https://github.com/MyRobotLab/myrobotlab/issues/294

hairygael, 2018-10-09:
okay,  yes these 2 issues seem relevant to what I have encountered.

closed 2018-12-20 by moz4r