MyRobotLab/myrobotlab #307: Breaking AIML in fr-FR

hairygael, 2018-10-09:
Version 1.1.294 Nixie
Playing with the swingui I tried to switch from complete en-US to "ear fr-FR", "mouth Celine female fr-FR", "bot fr-FR". It worked!
But now whenever I try to do the same, robot says systematicaly "I have no answer for that".

In Manticore, I used to look in the cmd.exe window to find a hint why the AIML was leading to such answer, but in Nixie, the cmd.exe window doesn't display those informations anymore.
How do you guys look for those info now?
[image]

moz4r, 2018-10-09:
also maybe we need to sort buttons

moz4r, 2018-10-15:
I just reproduced this bug ( "I have no answer for that" ) by chance ... fixed > if topic is null, it break spacetime.

closed 2018-10-22 by moz4r