MyRobotLab/myrobotlab #317: MRL install is broken

moz4r, 2018-10-12:
**Describe the bug**

Install "all" is broken ...

**To Reproduce**
Fresh folder, Install ( just ) opencv.
Then restart after opencv install
Then restart  and "Install all"

get this : `problem during retrieve of org.myrobotlab#myrobotlab-core: java.lang.RuntimeException: Multiple artifacts of the module org.bytedeco.javacpp-presets#ffmpeg;4.0.1-1.4.2 are retrieved to the same file! Update the retrieve pattern  to fix this error.`

**Desktop (please complete the following information):**
 - OS: win64

supertick, 2018-10-12:
[image]
Hmmm .. I followed instructions - but got this

moz4r, 2018-10-12:
[myrobotlab.log](https://github.com/MyRobotLab/myrobotlab/files/2473356/myrobotlab.log)

supertick, 2018-10-12:
Can't reproduce yet ... still trying :P

wilcovantoorn, 2018-10-15:
I did full install and forgot to noworky but yes it did reproduce. I did click on the myrobotlab.jar file in Windows and it started. Did the install all and it ran until the same issue as TS.
I'm sorry I forgot to noworky.
Maybe because of the way I started, without java -jar and junk double clicked the jar file.

supertick, 2018-11-02:
I have seen this too - but I'm not sure if its a mrl problem or an issue of catching the maven central or other transitive dependencies in flux.  It would be nice to freeze transitive version because I can still see maven going out to see if there are newer versions based on the descriptions of 1st level dependency transitives.
(next time someone sees it please re-open with more info) .. 

There was a multi-install fix done recently perhaps that fixed this problem too ?

moz4r, 2018-11-05:
just tested the last, I still have the issue.
Looked at the pom, maybe a conflict between org.boofcv and TesseractOcr ? not sure ...

kwatters, 2018-12-31:
I just tested here with the latest.  I installed opencv first. then I restarted, and did an install all.  there were no problems.

closed 2018-12-31 by kwatters