MyRobotLab/myrobotlab #331: Wrong behavior

hairygael, 2018-11-02:
**Describe the bug**
During install of components via runtime, the pop up window says immediately "finished" and "exit", although it still is installing components.

**To Reproduce**
Steps to reproduce the behavior:
1. start mrl with a fresh install (1.1.955)
2. in runtime select to install inmoov component
3. see the install pop up window doing the strange behavior

**Expected behavior**
window should proceed the install, and propse to exit only when finished

**Desktop (please complete the following information):**
 - OS: windows 10 pro
 - Browser chrome
 - Version 1.1.955

supertick, 2018-11-02:
I've see the same thing.  I believe it happens when multiple installs get fired off at "near" the same time.  One will finish without errors, yet another in the background is still installing.  I fixed this by making the installs synchronized (submitted pr)

supertick, 2018-11-02:
fixed in pr

PR #332 (2018-11-04): #331 - installation issue
Confirmed that multiple "installs" can be going on simultaneously. No
marked install as synchronized in all repo implementations. closes #331
commit 430564d7 (2018-11-04): #331 - installation issue

Confirmed that multiple "installs" can be going on simultaneously. No
marked install as synchronized in all repo implementations. closes #331
closed 2018-11-02 by supertick