MyRobotLab/myrobotlab #335: SRAIX repeats the question

hairygael, 2018-11-03:
**Describe the bug**
If you ask programAB a question related to a sraix, the robot will repeat your question, the user might think, the robot wants to learn the answer, but if you tell the correct answer the robot says he will remember, but in fact, he doesn't store the answer as it used to.

**To Reproduce**
Steps to reproduce the behavior:
1. ask the robot "who is michael jackson"
2. the robot repeats "who is michael jackson"
3. say: "michael jackson is a singer"
4. the robot says "ok, I will remember michael jackson is a singer"
5.save the predicates
6 ask the robot "who is michael jackson"
7. the robot repeats "who is michael jackson"

**Expected behavior**
In version 1.0.2693, sraix didn't always work, because of pannous failure, but some would work, if you would ask "who is michael jackson" the robot would search Answers.com to give the correct answer.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - windows 10 pro
 - chrome
 - Version 1.1.955

moz4r, 2018-11-04:
waiting other things, maybe we can push some stuff from french chatbot related to wikidata service and learn tag

kwatters, 2018-11-04:
This is my bug.  I'll fix it.  I left things a little 1/2 finished prior to the conference.  In MyRobotLab we have a class called  "MrlSraixHandler" ... This is where the bug is.  We override the SRAIX behavior from standard ProgramAB.   I'll try to preserve existing behavior, but it does remind us that we should avoid using Panous/pandorabots as a call out here.  We build a wikipedia index that is pretty decent at answering a lot of questions.  Only issue is we would need a cloud server that we could host that service on.  (these are longer term goals.. perhaps we could pay for the servers via patreon..)