MyRobotLab/myrobotlab #340: Clean up open source license headers in source files.

kwatters, 2018-11-13:
clean up to make sure all apache 2.0 license headers are proper and consistent across all mrl source files.  identify any license conficts.

LunDev, 2018-11-14:
My software engineering professor even had the opinion that there should be no license header etc. in each source file. Only once in the project root and for exceptions.
I don't care about which option is choosen, just this being consistent would be nice : )

moz4r, 2018-11-21:
I like the idea to put license Inside each files, if someone grab,only few functionalities from the project.
As in a cooked dish I would like to know where the meat comes from and the vegetables :)

kwatters, 2019-01-03:
We currently have the LICENSE file at the root.  I vote we remove all license files from source code that is not currently Apache 2.0.  If some file is non-apache 2.0 we can leave the license in place unless we're changing the license that it's under.