MyRobotLab/myrobotlab #349: Opencv framegrabbers noworky

moz4r, 2018-11-21:
**Describe the bug**

Other framegrabbers, than opencv framegrabber, no worky

**To Reproduce**
Steps to reproduce the behavior with sarxos or videoinput :
- Last build,
- Start opencv service
- Select  sarxos from combo
- Click capture
> camera is now on, but swingui throw error : could not get valid frame, buttun still indicate capture

```
14:26:59.851 [python] INFO c.m.s.Python [Python.java:475] exec(String) 
from org.myrobotlab.service import OpenCV
opencv = Runtime.getService("opencv")

14:27:00.549 [AWT-EventQueue-0] INFO c.m.f.Service [Service.java:2066] subscribe [opencv/publishStatus ---> gui/onStatus]
14:27:00.549 [AWT-EventQueue-0] INFO c.m.f.Service [Service.java:2066] subscribe [opencv/publishState ---> gui/onState]
14:27:15.279 [opencv-video-processor] INFO c.m.s.OpenCV [OpenCV.java:149] beginning capture
14:27:15.280 [opencv-video-processor] INFO c.m.s.OpenCV [OpenCV.java:760] video source is camera
14:27:15.282 [opencv-video-processor] INFO c.m.s.OpenCV [OpenCV.java:776] attempting to get frame grabber Sarxos format null
14:27:15.292 [opencv-video-processor] INFO c.m.s.OpenCV [OpenCV.java:808] using org.myrobotlab.opencv.SarxosFrameGrabber
14:27:15.317 [opencv-video-processor] INFO c.g.s.w.Webcam [Webcam.java:971] WebcamDefaultDriver capture driver will be used
14:27:15.852 [atomic-processor-1] INFO c.g.s.w.d.c.WebcamOpenTask [WebcamOpenTask.java:36] Opening webcam USB2.0 VGA UVC WebCam 0
14:27:19.300 [opencv-video-processor] INFO c.m.o.SarxosFrameGrabber [SarxosFrameGrabber.java:47] Camera Open With Resolution java.awt.Dimension[width=640,height=480]
```

supertick, 2018-11-24:
worky
[image]

hairygael, 2019-04-13:
Hello,
Testing version 1.1.157 with openCV
I get this error:
i01.opencv-video-processor-1] INFO  class org.myrobotlab.service.OpenCV - using org.bytedeco.javacv.OpenCVFrameGrabber
[ WARN:0] videoio(MSMF): OnReadSample() is called with error status: -1072875772
[ WARN:0] videoio(MSMF): async ReadSample() call is failed with error status: -1072875772
[ WARN:1] videoio(MSMF): can't grab frame. Error: -1072875772

supertick, 2019-04-13:
Gael, I need more information to reproduce the problem.  As you can see it currently worky for me.
[image]

hairygael, 2019-04-13:
Hello Man in Black,
I use the inmoov-develop with START_INMOOV.bat with the
service_D_OpenCv.config set as true.
C:\inmoov-develop\InMoov\config\service_D_OpenCv.config
The file inmoov_opencv.config contains choices regarding framegrabers and I
suspect things have changed in the last version which gets those choices
incompatible with mrl.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2019-04-14:
Ok, thanks .. I'll get there ..
First need to finish Servo refactoring then we can move onto InMoov

supertick, 2021-06-12:
closing ticket - as far as i can opencv frame grabbers worky - feel free to open if you find something that is opencv service related

closed 2021-06-12 by supertick