MyRobotLab/myrobotlab #363: OpenCV crash / issue while enable/disable yolo filter

moz4r, 2018-11-25:
**Describe the bug**
This sample is just to reproduce the issue.
Sometime runtime crash, sometime yolo filter is no reactivated

**To Reproduce**
Steps to reproduce the behavior:

```
opencv = Runtime.start("opencv", "OpenCV")
opencv.setGrabberType("OpenCV")
yoloFilter=opencv.addFilter("Yolo")
yoloFilter.disable()
opencv.capture()
yoloFilter.enable()
```

moz4r, 2018-11-25:
Tahnks !!! Can we imagine setActive() do enable() also ? Just to save I line code ...

supertick, 2018-11-27:
Fixed a very evil bug (created more multiple bug - do to the race condition) 
https://github.com/MyRobotLab/myrobotlab/commit/6e6dce45ae905ea766a84ed3e96e9dc8dea65208 

Tested with the following  script - (which is worky) and even though it sleeps for 2 seconds between enable/disable - it actually takes longer & blocks for the time for the OpenCVFilter to become ready for the next change ...

cv = Runtime.start("cv", "OpenCV")
cv.setGrabberType("OpenCV")
yoloFilter=cv.addFilter("Yolo")
yoloFilter.disable()
sleep(2)
cv.capture()
sleep(2)
yoloFilter.enable()
cv.setDisplayFilter("Yolo")
sleep(2)
yoloFilter.disable()
sleep(2)
yoloFilter.enable()
sleep(2)
yoloFilter.disable()
sleep(2)
yoloFilter.enable()