MyRobotLab/myrobotlab #365: organize the myrobotlab directory structure

kwatters, 2018-11-26:
When services install their dependencies/resources we would like them to have a consistent layout for the files.

Resources installed per service should be installed into a directory like

<mrl_install>/resources/ServiceType

Data that is generated by an instance of a service running should be stored in 

<mrl_install>/data/ServiceType/ServiceName

The service name sub directory can be optional in the case of data that is shared across all instances of a particular ServiceType.  (Example.  OpenCV cache.)

Log files generated from mrl running should be moved into the 

<mrl_install>/logs/  directory

finally, the .gitignore file should be cleaned up to include just 
data, resources, and logs

moz4r, 2018-11-26:
I like it, also this will cleanUp root folder.. nice, clean !
Compress logrotate at shutdown, maybe a good idea

supertick, 2022-02-06:
this was implemented

closed 2022-02-06 by supertick